bulletphysics/bullet3 #3123: deep mimic pybullet

bmahlbrand, 2020-10-27:
I followed the README and running the following seems to have no effect, no output, or GUI - could someone give advice on a quick start?

`python3 -m pybullet_envs.deep_mimic.testrl --arg_file run_humanoid3d_backflip_args.txt
`

erwincoumans, 2020-10-27:
It should work after pip3 install pybullet etc, tested on Windows, Linux and Mac OSX. Perhaps try on a different computer? Without output there is nothing we can do, we cannot reproduce it.

bmahlbrand, 2020-10-27:
Fair enough, managed to get it nearly running - but encountering the following problem after a few (could be one frame, not sure) frames render:

```
(base) PS C:\Users\$USER> python -m pybullet_envs.deep_mimic.testrl --arg_file run_humanoid3d_backflip_args.txt
parentdir= C:\Users\$USER\miniconda3\lib\site-packages
2020-10-26 22:46:32.864089: I tensorflow/stream_executor/platform/default/dso_loader.cc:49] Successfully opened dynamic library cudart64_110.dll
pybullet build time: Oct 26 2020 22:12:11
run_humanoid3d_backflip_args.txt
enable_draw= True
Initialization strategy: InitializationStrategy.RANDOM
starting thread 0
started testThreads thread 0 with threadHandle 00000000000005E4
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
Version = 4.6.0 NVIDIA 456.43
Vendor = NVIDIA Corporation
Renderer = GeForce RTX 2070 with Max-Q Design/PCIe/SSE2
b3Printf: Selected demo: Physics Server
starting thread 0
started MotionThreads thread 0 with threadHandle 0000000000000994
MotionThreadFunc thread started
motion_file= data/motions/humanoid3d_backflip.txt
LOADING humanoid!
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: urdfdom: no axis elemement for Joint, defaulting to (1,0,0) axis
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: chest
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: urdfdom: no axis elemement for Joint, defaulting to (1,0,0) axis
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: neck
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: urdfdom: no axis elemement for Joint, defaulting to (1,0,0) axis
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: right_hip
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: urdfdom: no axis elemement for Joint, defaulting to (1,0,0) axis
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: right_ankle
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: urdfdom: no axis elemement for Joint, defaulting to (1,0,0) axis
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: right_shoulder
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: urdfdom: no axis elemement for Joint, defaulting to (1,0,0) axis
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: left_hip
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: urdfdom: no axis elemement for Joint, defaulting to (1,0,0) axis
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: left_ankle
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: urdfdom: no axis elemement for Joint, defaulting to (1,0,0) axis
[... 194 lines cut ...]
  File "\site-packages\tensorflow\python\ops\resource_variable_ops.py", line 565, in value
    return self._read_variable_op()
  File "\site-packages\tensorflow\python\ops\resource_variable_ops.py", line 672, in _read_variable_op
    result = read_and_set_handle()
  File "\site-packages\tensorflow\python\ops\resource_variable_ops.py", line 662, in read_and_set_handle
    result = gen_resource_variable_ops.read_variable_op(
  File "\site-packages\tensorflow\python\ops\gen_resource_variable_ops.py", line 483, in read_variable_op
    _, _, _op, _outputs = _op_def_library._apply_op_helper(
  File "\site-packages\tensorflow\python\framework\op_def_library.py", line 748, in _apply_op_helper
    op = g._create_op_internal(op_type_name, inputs, dtypes=None,
  File "\site-packages\tensorflow\python\framework\ops.py", line 3528, in _create_op_internal
    ret = Operation(
  File "\site-packages\tensorflow\python\framework\ops.py", line 1990, in __init__
    self._traceback = tf_stack.extract_stack()

numActiveThreads = 0
stopping threads
Thread with taskId 0 with handle 0000000000000994 exiting
Thread TERMINATED
finished
numActiveThreads = 0
btShutDownExampleBrowser stopping threads
Thread with taskId 0 with handle 00000000000005E4 exiting
Thread TERMINATED
```

erwincoumans, 2020-10-27:
you need the right versions of all libs, and not the gpu version of tensorflow

bmahlbrand, 2020-10-27:
thanks, so for future readers, the issue is using tensorflow 2.3 - 2.4

closed 2020-10-27 by bmahlbrand