bulletphysics/bullet3 #3130: Joint acceleration limits in PyBullet

TimSchneider42, 2020-10-29:
Hey,

in a project I am currently working on, we are training a policy on the UR10 robot arm in simulation and transfer that policy afterwards to the real system. For that reason, the simulation has to be as accurate as possible to reduce divergence between simulation and reality to a minimum. 

During testing we figured out that the main source of error comes from the real robot controller limiting the joint acceleration (we are controlling the robot in velocity mode). Is there any way of limiting the joint accelerations to a fixed value in PyBullet? One workaround I could think of is to simply set the maximum velocity at each step accordingly. The problem with that is, that it will not enforce the acceleration limits during simulator substeps.

If there is no way of configuring that currently, could you hint me where in the bullet code such a change could be included? I suspect somewhere in the motor control code but I am having a hard time finding my way around in that large code base.

Thanks in advance.

Best,
Tim

erwincoumans, 2020-10-29:
Hi Tim,

There is a maxVelocity term in the setJointControlMode2 that you could use.

TimSchneider42, 2020-10-29:
Hi Erwin,

the only problem I saw with that is that I configured PyBullet to make 10 substeps each time I call stepSimulation. If I then specify the maximum velocity accordingly, I cannot guarantee that the acceleration limit is kept at each of the substeps, as the controller is likely going to accelerate a lot in the first substeps, then hit the velocity limit and afterwards maintains a constant velocity.

However, I did a quick comparison and figured out that in my scenario, calling stepSimulation 10 times with 1 substep configured is not substantially slower than calling stepSimulation 1 time with 10 substeps configured. So I am going to do that and use the maxVelocity term.

Thank your very much!

Best,
Tim

closed 2020-10-29 by TimSchneider42

ma-rv, 2020-11-19:
Hi @erwincoumans,

I noticed that panda joint accelerations are exceeding the limits specified in https://frankaemika.github.io/docs/control_parameters.html .
So is there any possibility to limit joint accelerations directly or would I need to implement a limit avoidance on the control side? I am using torque control. Position, velocity and torque limits are set.

Would be thankful for any help!