bulletphysics/bullet3 #3131: Redirection of stdout/stderr in pybullet

taiya, 2020-10-30:
Would be great if pybullet would not (unconditionally) pollute the standard streams. Had to resort to this hack to silence the `pybullet build time: Oct  8 2020 00:10:46` that was sent to stdout upon `import pybullet as pb`

```
# hide diagnostic output
with open(os.devnull, 'w') as devnull:
  # suppress stdout
  orig_stdout_fno = os.dup(sys.stdout.fileno())
  os.dup2(devnull.fileno(), 1)
  # suppress stderr
  orig_stderr_fno = os.dup(sys.stderr.fileno())
  os.dup2(devnull.fileno(), 2)
  import pybullet as pb
  os.fsync(devnull.fileno())
  os.dup2(orig_stdout_fno, 1)  # restore stdout
  os.dup2(orig_stderr_fno, 2)  # restore stderr
```

This was the only solution that worked out of what listed here (with an extra fsync needed):
https://stackoverflow.com/questions/4178614/suppressing-output-of-module-calling-outside-library
https://stackoverflow.com/questions/42952623/stop-python-module-from-printing

Would have been much nicer if the library would either 
1) respect the python sys.stdout / sys.stderr specification
2) allow to specify a different stream with a function

taiya, 2020-10-30:
Interestingly the fragment above only works when both stdout and stderr are redirected:
```
# hide diagnostic output
with open(os.devnull, 'w') as devnull:
  # suppress stdout
  orig_stdout_fno = os.dup(sys.stdout.fileno())
  os.dup2(devnull.fileno(), 1)
  import pybullet as pb
  os.fsync(devnull.fileno())
  os.dup2(orig_stdout_fno, 1)  # restore stdout
```

annoyingly results in:
```
Traceback (most recent call last):
  File "/workspaces/kubric/helloworld.py", line 44, in <module>
    os.fsync(devnull.fileno())
OSError: [Errno 22] Invalid argument
```

taiya, 2020-10-30:
A more cleaned up version, including a potential leak prevention caused by dup.

```
class RedirectStream(object):

  @staticmethod
  def _flush_c_stream():
    libc = ctypes.CDLL(None)
    libc.fflush(ctypes.c_void_p.in_dll(libc, 'stdout'))

  def __init__(self, stream=sys.stdout, file=os.devnull):
    self.stream = stream
    self.file = file

  def __enter__(self):
    self.stream.flush()  # ensures python stream unaffected 
    self.fd = open(self.file, "w")
    self.dup_stream = os.dup(self.stream.fileno())
    os.dup2(self.fd.fileno(), self.stream.fileno()) # replaces stream
  
  def __exit__(self, type, value, traceback):
    RedirectStream._flush_c_stream()  # ensures C stream buffer empty
    os.dup2(self.dup_stream, self.stream.fileno()) # restores stream
    os.close(self.dup_stream)
    self.fd.close()

with RedirectStream(sys.stdout):
  import pybullet as pb
```

erwincoumans, 2020-11-03:
Thanks, didn't realize anyone case about the output messages. Some of the printfs go through a common C++ function, so it would be possible to clean up the code (and some people don't like the build time output etc).
I'll leave it open for now, hope to get to this, since the project severely lacks resources.

ZhiquanW, 2020-11-06:
I hope that pybullet can provide a parameter to control the output of these annoying messages as the outputs can be a mess while running with multiprocess.

closed 2021-04-26 by erwincoumans