bulletphysics/bullet3 #3134: ffmpeg 4.0 incompatiblity

LemonPi, 2020-11-03:
I'm having trouble state logging to video with:
```python
p.startStateLogging(p.STATE_LOGGING_VIDEO_MP4, "video.mp4")
```
after either upgrading `ffmpeg` and `pybullet`. Their current versions are:
```
pybullet: 2.7.9
ffmpeg: 4.0
```

The error log I'm getting is:
```
Backend Qt5Agg is interactive backend. Turning interactive mode on.
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
Creating context
Created GL 3.3 context
Direct GLX rendering context obtained
Making context current
GL_VENDOR=NVIDIA Corporation
GL_RENDERER=GeForce GTX 1080 Ti/PCIe/SSE2
GL_VERSION=3.3.0 NVIDIA 396.18
GL_SHADING_LANGUAGE_VERSION=3.30 NVIDIA via Cg compiler
pthread_getconcurrency()=0
Version = 3.3.0 NVIDIA 396.18
Vendor = NVIDIA Corporation
Renderer = GeForce GTX 1080 Ti/PCIe/SSE2
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0 
MotionThreadFunc thread started
ven = NVIDIA Corporation
ven = NVIDIA Corporation
ffmpeg version 4.0 Copyright (c) 2000-2018 the FFmpeg developers
  built with gcc 7.2.0 (crosstool-NG fa8859cb)
  configuration: --prefix=/opt/conda/conda-bld/ffmpeg_1531088893642/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placeh --cc=/opt/conda/conda-bld/ffmpeg_1531088893642/_build_env/bin/x86_64-conda_cos6-linux-gnu-cc --disable-doc --enable-shared --enable-static --enable-zlib --enable-pic --enable-gpl --enable-version3 --disable-nonfree --enable-hardcoded-tables --enable-avresample --enable-libfreetype --disable-openssl --disable-gnutls --enable-libvpx --enable-pthreads --enable-libopus --enable-postproc --disable-libx264
  libavutil      56. 14.100 / 56. 14.100
  libavcodec     58. 18.100 / 58. 18.100
  libavformat    58. 12.100 / 58. 12.100
  libavdevice    58.  3.100 / 58.  3.100
  libavfilter     7. 16.100 /  7. 16.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  1.100 /  5.  1.100
  libswresample   3.  1.100 /  3.  1.100
  libpostproc    55.  1.100 / 55.  1.100
Unrecognized option 'preset'.
Error splitting the argument list: Option not found
```

This is in Ubuntu 16.04 if that makes a difference.

erwincoumans, 2020-11-03:
Can you try using a recent ffmpeg?

I tried todays release version and it works just fine:
~~~~~~~~~~~~~~~~~~~
D:\dev\bullet3\examples\pybullet\examples>python video_sync_mp4.py
pybullet build time: Oct 27 2020 08:20:45
argv[0]=--width=1024
argv[1]=--height=768
argv[2]=--mp4="test.mp4"
argv[3]=--mp4fps=240
starting thread 0
started testThreads thread 0 with threadHandle 0000000000000240
argc=6
argv[0] = --unused
argv[1] = --width=1024
argv[2] = --height=768
argv[3] = --mp4="test.mp4"
argv[4] = --mp4fps=240
argv[5] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
Version = 4.6.0 NVIDIA 446.14
Vendor = NVIDIA Corporation
Renderer = GeForce RTX 2080/PCIe/SSE2
b3Printf: Selected demo: Physics Server
starting thread 0
started MotionThreads thread 0 with threadHandle 000000000000053C
MotionThreadFunc thread started
ffmpeg version N-99760-gecd379087b-ge5be4c5a88+2 Copyright (c) 2000-2020 the FFmpeg developers
  built with gcc 10.2.0 (Rev4, Built by MSYS2 project)
  configuration:  --cc='ccache gcc' --cxx='ccache g++' --disable-autodetect --enable-amf --enable-bzlib --enable-cuda --enable-cuvid --enable-d3d11va --enable-dxva2 --enable-iconv --enable-lzma --enable-nvenc --enable-zlib --enable-sdl2 --enable-ffnvcodec --enable-nvdec --enable-cuda-llvm --enable-libmp3lame --enable-libopus --enable-libvorbis --enable-libvpx --enable-libx264 --enable-libx265 --enable-libdav1d --enable-libaom --disable-debug --enable-fontconfig --enable-libass --enable-libbluray --enable-libfreetype --enable-libmfx --enable-libmysofa --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopenjpeg --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvo-amrwbenc --enable-libwebp --enable-libxml2 --enable-libzimg --enable-libshine --enable-gpl --enable-avisynth --enable-libxvid --enable-libopenmpt --enable-version3 --enable-librav1e --enable-libsrt --enable-libgsm --enable-libvmaf --enable-libsvtav1 --enable-librtmp --enable-mbedtls --extra-cflags=-DLIBTWOLAME_STATIC --extra-libs=-lstdc++ --extra-cflags=-DLIBXML_STATIC --extra-libs=-liconv --disable-w32threads
  libavutil      56. 60.100 / 56. 60.100
  libavcodec     58.112.100 / 58.112.100
  libavformat    58. 63.100 / 58. 63.100
  libavdevice    58. 11.102 / 58. 11.102
  libavfilter     7. 88.100 /  7. 88.100
  libswscale      5.  8.100 /  5.  8.100
  libswresample   3.  8.100 /  3.  8.100
  libpostproc    55.  8.100 / 55.  8.100
[rawvideo @ 000002068d1ce8c0] Stream #0: not enough frames to estimate rate; consider increasing probesize
Input #0, rawvideo, from 'pipe:':
  Duration: N/A, start: 0.000000, bitrate: 6039797 kb/s
    Stream #0:0: Video: rawvideo (RGBA / 0x41424752), rgba, 1024x768, 6039797 kb/s, 240 tbr, 240 tbn, 240 tbc
Stream mapping:
  Stream #0:0 -> #0:0 (rawvideo (native) -> h264 (libx264))
[libx264 @ 000002068d1e3680] using cpu capabilities: MMX2 SSE2Fast SSSE3 SSE4.2 AVX FMA3 BMI2 AVX2
[libx264 @ 000002068d1e3680] profile High, level 5.1, 4:2:0, 8-bit
[libx264 @ 000002068d1e3680] 264 - core 161 r3027 4121277 - H.264/MPEG-4 AVC codec - Copyleft 2003-2020 - http://www.videolan.org/x264.html - options: cabac=1 ref=5 deblock=1:0:0 analyse=0x3:0x113 me=hex subme=8 psy=1 psy_rd=1.00:0.00 mixed_ref=1 me_range=16 chroma_me=1 trellis=2 8x8dct=1 cqm=0 deadzone=21,11 fast_pskip=1 chroma_qp_offset=-2 threads=24 lookahead_threads=4 sliced_threads=0 nr=0 decimate=1 interlaced=0 bluray_compat=0 constrained_intra=0 bframes=3 b_pyramid=2 b_adapt=1 b_bias=0 direct=3 weightb=1 open_gop=0 weightp=2 keyint=250 keyint_min=25 scenecut=40 intra_refresh=0 rc_lookahead=50 rc=crf mbtree=1 crf=22.0 qcomp=0.60 qpmin=0 qpmax=69 qpstep=4 ip_ratio=1.40 aq=1:1.00
Output #0, mp4, to 'test.mp4':
  Metadata:
    encoder         : Lavf58.63.100
    Stream #0:0: Video: h264 (libx264) (avc1 / 0x31637661), yuv420p, 1024x768, q=-1--1, 50000 kb/s, 240 fps, 15360 tbn, 240 tbc
    Metadata:
      encoder         : Lavc58.112.100 libx264
    Side data:
      cpb: bitrate max/min/avg: 0/0/50000000 buffer size: 0 vbv_delay: N/A
numActiveThreads = 0q=32.0 size=       0kB time=00:00:02.65 bitrate=   0.1kbits/s speed=0.156x
stopping threads
Thread with taskId 0 with handle 000000000000053C exiting
Thread TERMINATED
finished
numActiveThreads = 0
btShutDownExampleBrowser stopping threads
Thread with taskId 0 with handle 0000000000000240 exiting
Thread TERMINATED
frame=  721 fps= 42 q=-1.0 Lsize=     201kB time=00:00:02.99 bitrate= 549.7kbits/s speed=0.174x
video:191kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 4.833980%
[libx264 @ 000002068d1e3680] frame I:4     Avg QP:16.53  size: 34811
[libx264 @ 000002068d1e3680] frame P:182   Avg QP:22.69  size:   113
[libx264 @ 000002068d1e3680] frame B:535   Avg QP:27.22  size:    66
[libx264 @ 000002068d1e3680] consecutive B-frames:  0.8%  0.3%  1.2% 97.6%
[libx264 @ 000002068d1e3680] mb I  I16..4: 53.4% 18.6% 28.0%
[libx264 @ 000002068d1e3680] mb P  I16..4:  0.0%  0.0%  0.0%  P16..4:  0.1%  0.1%  0.0%  0.0%  0.0%    skip:99.7%
[libx264 @ 000002068d1e3680] mb B  I16..4:  0.0%  0.0%  0.0%  B16..8:  0.2%  0.0%  0.0%  direct: 0.0%  skip:99.7%  L0:40.1% L1:58.0% BI: 1.9%
[libx264 @ 000002068d1e3680] 8x8 transform intra:19.9% inter:14.4%
[libx264 @ 000002068d1e3680] direct mvs  spatial:97.6% temporal:2.4%
[libx264 @ 000002068d1e3680] coded y,uvDC,uvAC intra: 20.1% 31.0% 27.9% inter: 0.0% 0.0% 0.0%
[... cut ...]
[libx264 @ 000002068d1e3680] i8 v,h,dc,ddl,ddr,vr,hd,vl,hu: 45% 15% 29%  2%  2%  1%  2%  1%  3%
[libx264 @ 000002068d1e3680] i4 v,h,dc,ddl,ddr,vr,hd,vl,hu: 10% 18% 26%  5%  5%  3%  8%  6% 19%
[libx264 @ 000002068d1e3680] i8c dc,h,v,p: 82% 12%  5%  2%
[libx264 @ 000002068d1e3680] Weighted P-Frames: Y:0.0% UV:0.0%
[libx264 @ 000002068d1e3680] ref P L0: 58.4%  2.9% 10.4%  6.4% 20.8%  1.1%
[libx264 @ 000002068d1e3680] ref B L0: 67.6% 21.9%  9.3%  1.2%
[libx264 @ 000002068d1e3680] ref B L1: 93.9%  6.1%
[libx264 @ 000002068d1e3680] kb/s:520.35

~~~~~~~~~~~~~~~~~~~

closed 2020-11-03 by erwincoumans

LemonPi, 2020-11-03:
Works with ffmpeg 4.3.1; thanks!