bulletphysics/bullet3 #3135: saveUrdf doesn't save <limits> for joints of type revolute

AchkanSalehi, 2020-11-03:
Consider this code snippet:

    import time
    import sys
    import pybullet as p
    from pybullet_utils import urdfEditor
   
    p.connect(p.DIRECT)
    p.loadURDF(sys.argv[1])

    ue=urdfEditor.UrdfEditor()
    ue.initializeFromBulletBody(0,0)
    ue.saveUrdf("/tmp/same_urdf.urdf")

If you pass a `*urdf` file to this script that contains a joint of type `revolute` with the `<limit>` fields,  _e.g._

    
    <joint name="left_gripper_joint" type="revolute">
        <axis xyz="0 0 1"/>
        <limit effort="1000.0" lower="0.0" upper="0.548" velocity="0.5"/>
        <origin rpy="0 0 0" xyz="0.2 0.01 0"/>
        <parent link="gripper_pole"/>
        <child link="left_gripper"/>
    </joint>

then those `<limit>` fields wont be included in the file that results from calling `saveUrdf` 

    
    <joint name="left_gripper_joint" type="revolute">
        <parent link="gripper_pole"/>
        <child link="left_gripper"/>
        <dynamics damping="1.0" friction="0.0001"/>
        <origin rpy="0.00000 -0.00000 0.00000" xyz="0.20000 0.01000 0.00000"/>
        <axis xyz="0.00000 0.00000 1.00000"/>
    </joint>

and it also adds another field `<dynamics>`.

Is that a bug or am I just missing something obvious?

erwincoumans, 2020-11-03:
The URDF editor is a starting point indeed. Since the source code is available in Python, you can inspect what is supported and not, and add any features you want or need.

closed 2020-11-03 by erwincoumans

AchkanSalehi, 2020-11-03:
Fair enough.