bulletphysics/bullet3 #3137: URDF files not found any more

florianHofherr, 2020-11-04:
Hi everyone,

I am experiencing the following problem: I have installed the C++ Bullet Physics SDK using vcpkg and built pybullet for my environment using 'python setup.py build' and 'python setup.py install'. Initially everything works fine, and the sample code, e.g. in `deformable_torus.py` does what it should. 

However, at some point, I start getting the error, that the URDF files are not found any more. When I uninstall pybullet and re-install it (without changing anything else) then the URDF files are found again. 

Does anyone have an idea what the problem might be? I have done this cycle a few times now, but I do not yet see a pattern on when the URDF files start not getting found any more.

This is the error message which I get:

```
pybullet build time: Nov  2 2020 09:37:49
starting thread 0
started testThreads thread 0 with threadHandle 000000000000021C
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
Version = 4.6.13596 Compatibility Profile Context 20.10.28 27.20.1028.1
Vendor = ATI Technologies Inc.    
Renderer = AMD Radeon(TM) Graphics
b3Printf: Selected demo: Physics Server
starting thread 0
started MotionThreads thread 0 with threadHandle 00000000000005D8
MotionThreadFunc thread started
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,152]:

b3Printf: URDF file 'plane.urdf' not found

Traceback (most recent call last):
  File "c:/dev/vcpkg/buildtrees/bullet3/src/2.89-eb26871ed5.clean/examples/pybullet/examples/deformable_torus.py", line 10, 
in <module>
    planeId = p.loadURDF("plane.urdf", [0,0,-2])
pybullet.error: Cannot load URDF file.
numActiveThreads = 0
stopping threads
Thread with taskId 0 with handle 00000000000005D8 exiting
Thread TERMINATED
finished
numActiveThreads = 0
btShutDownExampleBrowser stopping threads
Thread with taskId 0 with handle 000000000000021C exiting
Thread TERMINATED
```

Thanks for your help!

florianHofherr, 2020-11-04:
Found the problem, sorry.

closed 2020-11-04 by florianHofherr

huzheyuan740, 2021-03-05:
Hello! How did you uninstall the pybullet installed with CMake?  My environment is Linux.