bulletphysics/bullet3 #3140: Batch createMultiBody returns non-unique ids after removeBody

rmsc, 2020-11-06:
I'm periodically creating batches of few hundreds of objects with `createMultiBody`, and also periodically remove a few of them using `removeBody`. After removing objects from the world, some of the ids returned by `crateMultiBody` are actually colliding with ids of existing objects.

I've noticed that the returned ids are contiguous and seem to start at `len(existing_bodies)`. This obviously breaks if random objects are removed.

This is pybullet 3.05.

rmsc, 2020-11-06:
Indeed that seems to be the case:

```
int uid = b3GetStatusBodyIndex(statusHandle);
if (numBatchPositions > 0)
{
        PyObject* pyResultList = PyTuple_New(numBatchPositions );
        for (i = 0; i < numBatchPositions; i++)
        {
                PyTuple_SetItem(pyResultList, i, PyLong_FromLong(uid - numBatchPositions + i + 1));
        }
        return pyResultList;
}
else
{
        PyObject* ob = PyLong_FromLong(uid);
        return ob;
}
```

erwincoumans, 2020-11-06:
Thanks for the report. Can you create a small repro case/test?

rmsc, 2020-11-06:
Sure, here's a small test case:
```
import pybullet as pb

pb.connect(pb.GUI)
vis = pb.createVisualShape(shapeType = pb.GEOM_SPHERE, radius = 1.0)
col = pb.createCollisionShape(shapeType = pb.GEOM_SPHERE, radius = 1.0)

def create_3_bodies():
    positions = [[ 0.0, 0.0, 0.0],
                 [ 1.0, 0.0, 0.0],
                 [ 2.0, 0.0, 0.0]]
    return pb.createMultiBody(baseMass = 1.0,
                              baseCollisionShapeIndex = col,
                              baseVisualShapeIndex = vis,
                              batchPositions = positions,
                              basePosition = [0, 0, 0],
                              baseOrientation = pb.getQuaternionFromEuler([0, 0, 0]))

bodies = []
# create 3 bodies
bodies += create_3_bodies()
print(bodies)
# [0, 1, 2]

# remove first body
pb.removeBody(bodies[0])
bodies.pop(0)
print(bodies)
# [1, 2]

# create additional bodies
bodies += create_3_bodies()
print(bodies)
# [1, 2, 2, 3, 4]
#     ^--^--repeated IDS!
```

erwincoumans, 2020-11-06:
Thanks for the repro case, it is a but for batch creation indeed, will have a look later.

closed 2021-04-26 by erwincoumans