bulletphysics/bullet3 #3142: Unexpected behavior with bullet engine.

lshachar, 2020-11-07:
Hi!
I'm using bullet physics in a robotics simulator called CoppeliaSim (Previously named V-rep).
I ran into a very unexpected behavior, that gets corrected by using a different physics engine. (the simulator supports several: ODE, Newton, Vortex, Bullet. I just switched to ODE/Newton.)
I asked CoppeliaSim's devs for a response and they say that all physics engines are handled identically internally, which further reduces the chances that the bug is in the simulation software. (btw, it's open source)

[I set up a scene in the simulator which specifically shows the bug.](https://forum.coppeliarobotics.com/viewtopic.php?f=5&t=8892) CoppeliaSim can be run on Windows / Mac / Linux, However I think reading my bug description (in the provided link) should be enough for a bullet developer to set up a similar suitable test.

erwincoumans, 2020-11-13:
We would need a reproduction case. What constraint type is used exactly for the hinge?
If you can reproduce this in Bullet C++ or PyBullet we can fix it.

lshachar, 2020-11-19:
Hello @erwincoumans,
The top cube is connected to the sliding cube by a prismatic joint. the joint's position is controlled by a PID loop. I never used Bullet C++ or PyBullet, so I can't help much there. but I feel pretty confident that if anyone [follows my recipe](https://forum.coppeliarobotics.com/viewtopic.php?f=5&t=8892) the buggy behavior will emerge. I can make an explanation video, to illustrate the wrong behavior better. Please let me know if I can help.

erwincoumans, 2020-11-19:
We need to know the C++ constraint type to reproduce: btHingeConstraint? btGeneric6DofConstraint? btGeneric6DofSpring2Constraint?

lshachar, 2020-11-22:
Coppelia's response:
'This is a btSliderConstraint object.'

I hope this helps!

closed 2021-04-26 by erwincoumans