bulletphysics/bullet3 #3146: The contact points have been not removed cleanly

WinsonDeng, 2020-11-13:
Hello everyone,

I create a btPairCachingGhostObject object as sensor and a btRigidBody object.
Then, I control the rigid body to move along X-axis (move 0.005m each time) until colliding with sensor and then passing through it.

A illogical issue has been found: when the rigid body passes through the sensor from left to right, the contact points have been not removed cleanly. I'm sure that two objects are separated.

Is it something wrong with Bullet collision detection algorithm?
It can be sure that btBoxBoxCollisionAlgorithm doesn't compute correct contact point loction and normal vector, which causing penetration distance is still negative when two objects are separated.

Please check the forum post for details: 
[https://pybullet.org/Bullet/phpBB3/viewtopic.php?f=9&t=13106](url)

[https://i.loli.net/2020/11/12/H1zucA6CxF3ZqOy.jpg](url)

[https://i.loli.net/2020/11/12/wyFlB4hbxKL3AM2.jpg](url)

erwincoumans, 2020-11-13:
Where is stepSimulation or performDiscreteCollisionDetection called, to compute/refresh contact points?

WinsonDeng, 2020-11-13:
My test code is based on [BasicExample.cpp](https://github.com/bulletphysics/bullet3/blob/master/examples/BasicDemo/BasicExample.cpp) which inheritance [CommonRigidBodyBase.h](https://github.com/bulletphysics/bullet3/blob/master/examples/CommonInterfaces/CommonRigidBodyBase.h)
I don't know where is `CommonRigidBodyBase::stepSimulation(deltaTime)` called.

erwincoumans, 2020-11-13:
Thanks for the report.
The contact points are cached, and this doesn't work well for ghost objects, you need to clear the contact points before each simulation step. For example using a pre tick callback, after you created the sensorGhost:
~~~~~~~~~~~~~~~~
m_dynamicsWorld->setInternalTickCallback(tickCallback, sensorGhost, true);
~~~~~~~~~~~~~~~~

~~~~~~~~~~~~~~~~~
void tickCallback(btDynamicsWorld* world, btScalar timeStep)
{
	void* sensorGhost = world->getWorldUserInfo();
	for (int i = 0; i < world->getDispatcher()->getNumManifolds(); i++)
	{
		btPersistentManifold* contactManifold = world->getDispatcher()->getManifoldByIndexInternal(i);
		if (contactManifold->getBody0() == sensorGhost || contactManifold->getBody1() == sensorGhost)
			contactManifold->clearManifold();
	}
}
~~~~~~~~~~~~~~~~~

closed 2020-11-13 by erwincoumans