bulletphysics/bullet3 #3148: Invalid keyword when using Pybullet.calculateInverseKinematics(axNumIterations=100, residualThreshold=0.01)

YuZhan69, 2020-11-14:
I try to use pybullet==2.8.4 in win10 virtual env and implemented         
target_joint_state = p.calculateInverseKinematics(bodyIndex=self.robot_body_id,
                                                          endEffectorLinkIndex=self.robot_end_effector_link_index,
                                                          targetPosition=position, targetOrientation=orientation,
                                                          maxNumIterations=100, residualThreshold=0.01),
got an error as invalid keyword argument for this function for "maxNumIterations" & "residualThreshold". 
Also tried pybullet==3.0.6, not work either.
Could anyone help with that? Thanks!

erwincoumans, 2020-11-14:
Please check inverse_kinematics.py example, and set useNullspace=0
https://github.com/erwincoumans/bullet3/blob/master/examples/pybullet/examples/inverse_kinematics.py

closed 2020-11-14 by erwincoumans

YuZhan69, 2020-11-14:
Thank you very much!