bulletphysics/bullet3 #3152: bullet 3.06 build issue

chenrui333, 2020-11-18:
👋 trying to build the latest release, but run into some build issue. The error log is as below:

<details>
<summary>build failure</summary>

```
==> cmake .. -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -DCMAKE_INSTALL_PREFIX=/usr/local/Cellar/bullet/3.06 -DCMAKE_BUILD_TYPE=Release -DCMAKE_FIND_FRAMEWORK=LAST -DCMAKE_VERBOSE_MAKEFILE=ON -Wno-dev -DCMAKE_OSX_SYSROOT=/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk -DBUILD_BULLET2_DEMOS=OFF -DBUILD_PYBULLET=OFF -DBUILD_UNIT_TESTS=OFF -DCMAKE_INSTALL_RPATH=/usr/local/Cellar/bullet/3.06/lib -DINSTALL_EXTRA_LIBS=ON -DBUILD_SHARED_LIBS=ON
-- The C compiler identification is AppleClang 12.0.0.12000032
-- The CXX compiler identification is AppleClang 12.0.0.12000032
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang - skipped
[ 99%] Building CXX object Extras/BulletRoboticsGUI/CMakeFiles/BulletRoboticsGUI.dir/__/__/examples/SharedMemory/PhysicsServerExample.o
cd /tmp/bullet-20201113-27819-1imajac/bullet3-3.06/build/Extras/BulletRoboticsGUI && /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++ -DBT_ENABLE_CLSOCKET -DBT_ENABLE_ENET -DBulletRoboticsGUI_EXPORTS -DHAS_SOCKLEN_T -DPHYSICS_SERVER_DIRECT -DUSE_GRAPHICAL_BENCHMARK -D_DARWIN -I/tmp/bullet-20201113-27819-1imajac/bullet3-3.06/src -I/tmp/bullet-20201113-27819-1imajac/bullet3-3.06/examples -I/tmp/bullet-20201113-27819-1imajac/bullet3-3.06/examples/SharedMemory -I/tmp/bullet-20201113-27819-1imajac/bullet3-3.06/examples/ThirdPartyLibs -I/tmp/bullet-20201113-27819-1imajac/bullet3-3.06/examples/ThirdPartyLibs/enet/include -I/tmp/bullet-20201113-27819-1imajac/bullet3-3.06/examples/ThirdPartyLibs/clsocket/src -DNDEBUG -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk -fPIC -o CMakeFiles/BulletRoboticsGUI.dir/__/__/examples/SharedMemory/PhysicsServerExample.o -c /tmp/bullet-20201113-27819-1imajac/bullet3-3.06/examples/SharedMemory/PhysicsServerExample.cpp
[ 99%] Building CXX object Extras/BulletRoboticsGUI/CMakeFiles/BulletRoboticsGUI.dir/__/__/examples/SharedMemory/PhysicsServerExampleBullet2.o
cd /tmp/bullet-20201113-27819-1imajac/bullet3-3.06/build/Extras/BulletRoboticsGUI && /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++ -DBT_ENABLE_CLSOCKET -DBT_ENABLE_ENET -DBulletRoboticsGUI_EXPORTS -DHAS_SOCKLEN_T -DPHYSICS_SERVER_DIRECT -DUSE_GRAPHICAL_BENCHMARK -D_DARWIN -I/tmp/bullet-20201113-27819-1imajac/bullet3-3.06/src -I/tmp/bullet-20201113-27819-1imajac/bullet3-3.06/examples -I/tmp/bullet-20201113-27819-1imajac/bullet3-3.06/examples/SharedMemory -I/tmp/bullet-20201113-27819-1imajac/bullet3-3.06/examples/ThirdPartyLibs -I/tmp/bullet-20201113-27819-1imajac/bullet3-3.06/examples/ThirdPartyLibs/enet/include -I/tmp/bullet-20201113-27819-1imajac/bullet3-3.06/examples/ThirdPartyLibs/clsocket/src -DNDEBUG -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk -fPIC -o CMakeFiles/BulletRoboticsGUI.dir/__/__/examples/SharedMemory/PhysicsServerExampleBullet2.o -c /tmp/bullet-20201113-27819-1imajac/bullet3-3.06/examples/SharedMemory/PhysicsServerExampleBullet2.cpp
[100%] Building CXX object Extras/BulletRoboticsGUI/CMakeFiles/BulletRoboticsGUI.dir/__/__/examples/SharedMemory/SharedMemoryInProcessPhysicsC_API.o
cd /tmp/bullet-20201113-27819-1imajac/bullet3-3.06/build/Extras/BulletRoboticsGUI && /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++ -DBT_ENABLE_CLSOCKET -DBT_ENABLE_ENET -DBulletRoboticsGUI_EXPORTS -DHAS_SOCKLEN_T -DPHYSICS_SERVER_DIRECT -DUSE_GRAPHICAL_BENCHMARK -D_DARWIN -I/tmp/bullet-20201113-27819-1imajac/bullet3-3.06/src -I/tmp/bullet-20201113-27819-1imajac/bullet3-3.06/examples -I/tmp/bullet-20201113-27819-1imajac/bullet3-3.06/examples/SharedMemory -I/tmp/bullet-20201113-27819-1imajac/bullet3-3.06/examples/ThirdPartyLibs -I/tmp/bullet-20201113-27819-1imajac/bullet3-3.06/examples/ThirdPartyLibs/enet/include -I/tmp/bullet-20201113-27819-1imajac/bullet3-3.06/examples/ThirdPartyLibs/clsocket/src -DNDEBUG -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk -fPIC -o CMakeFiles/BulletRoboticsGUI.dir/__/__/examples/SharedMemory/SharedMemoryInProcessPhysicsC_API.o -c /tmp/bullet-20201113-27819-1imajac/bullet3-3.06/examples/SharedMemory/SharedMemoryInProcessPhysicsC_API.cpp
[100%] Linking CXX shared library libBulletRoboticsGUI.dylib
cd /tmp/bullet-20201113-27819-1imajac/bullet3-3.06/build/Extras/BulletRoboticsGUI && /usr/local/Cellar/cmake/3.18.4/bin/cmake -E cmake_link_script CMakeFiles/BulletRoboticsGUI.dir/link.txt --verbose=1
/usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++ -DNDEBUG -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk -dynamiclib -Wl,-headerpad_max_install_names -compatibility_version 3.5.0 -current_version 3.5.0 -o libBulletRoboticsGUI.3.05.dylib -install_name @rpath/libBulletRoboticsGUI.3.05.dylib CMakeFiles/BulletRoboticsGUI.dir/__/__/examples/SharedMemory/plugins/stablePDPlugin/SpAlg.o CMakeFiles/BulletRoboticsGUI.dir/__/__/examples/SharedMemory/plugins/stablePDPlugin/Shape.o CMakeFiles/BulletRoboticsGUI.dir/__/__/examples/SharedMemory/plugins/stablePDPlugin/RBDUtil.o CMakeFiles/BulletRoboticsGUI.dir/__/__/examples/SharedMemory/plugins/stablePDPlugin/RBDModel.o CMakeFiles/BulletRoboticsGUI.dir/__/__/examples/SharedMemory/plugins/stablePDPlugin/MathUtil.o CMakeFiles/BulletRoboticsGUI.dir/__/__/examples/SharedMemory/plugins/stablePDPlugin/KinTree.o CMakeFiles/BulletRoboticsGUI.dir/__/__/examples/SharedMemory/plugins/stablePDPlugin/BulletConversion.o CMakeFiles/BulletRoboticsGUI.dir/__/__/examples/ExampleBrowser/InProcessExampleBrowser.o CMakeFiles/BulletRoboticsGUI.dir/__/__/examples/SharedMemory/GraphicsServerExample.o CMakeFiles/BulletRoboticsGUI.dir/__/__/examples/SharedMemory/GraphicsClientExample.o CMakeFiles/BulletRoboticsGUI.dir/__/__/examples/SharedMemory/RemoteGUIHelper.o CMakeFiles/BulletRoboticsGUI.dir/__/__/examples/SharedMemory/RemoteGUIHelperTCP.o CMakeFiles/BulletRoboticsGUI.dir/__/__/examples/SharedMemory/PhysicsServerExample.o CMakeFiles/BulletRoboticsGUI.dir/__/__/examples/SharedMemory/PhysicsServerExampleBullet2.o CMakeFiles/BulletRoboticsGUI.dir/__/__/examples/SharedMemory/SharedMemoryInProcessPhysicsC_API.o  -Wl,-rpath,/tmp/bullet-20201113-27819-1imajac/bullet3-3.06/build/Extras/BulletRobotics -Wl,-rpath,/tmp/bullet-20201113-27819-1imajac/bullet3-3.06/build/Extras/InverseDynamics -Wl,-rpath,/tmp/bullet-20201113-27819-1imajac/bullet3-3.06/build/Extras/Serialize/BulletWorldImporter -Wl,-rpath,/tmp/bullet-20201113-27819-1imajac/bullet3-3.06/build/Extras/Serialize/BulletFileLoader -Wl,-rpath,/tmp/bullet-20201113-27819-1imajac/bullet3-3.06/build/src/BulletSoftBody -Wl,-rpath,/tmp/bullet-20201113-27819-1imajac/bullet3-3.06/build/src/BulletDynamics -Wl,-rpath,/tmp/bullet-20201113-27819-1imajac/bullet3-3.06/build/src/BulletCollision -Wl,-rpath,/tmp/bullet-20201113-27819-1imajac/bullet3-3.06/build/src/BulletInverseDynamics -Wl,-rpath,/tmp/bullet-20201113-27819-1imajac/bullet3-3.06/build/src/LinearMath -Wl,-rpath,/tmp/bullet-20201113-27819-1imajac/bullet3-3.06/build/src/Bullet3Common -lBulletExampleBrowserLib ../BulletRobotics/libBulletRobotics.3.05.dylib ../InverseDynamics/libBulletInverseDynamicsUtils.3.05.dylib ../Serialize/BulletWorldImporter/libBulletWorldImporter.3.05.dylib ../Serialize/BulletFileLoader/libBulletFileLoader.3.05.dylib ../../src/BulletSoftBody/libBulletSoftBody.3.05.dylib ../../src/BulletDynamics/libBulletDynamics.3.05.dylib ../../src/BulletCollision/libBulletCollision.3.05.dylib ../../src/BulletInverseDynamics/libBulletInverseDynamics.3.05.dylib ../../src/LinearMath/libLinearMath.3.05.dylib ../../src/Bullet3Common/libBullet3Common.3.05.dylib 
ld: library not found for -lBulletExampleBrowserLib
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [Extras/BulletRoboticsGUI/libBulletRoboticsGUI.3.05.dylib] Error 1
make[1]: *** [Extras/BulletRoboticsGUI/CMakeFiles/BulletRoboticsGUI.dir/all] Error 2
make: *** [all] Error 2

```

</details>

Full build log is here, https://github.com/Homebrew/homebrew-core/runs/1394359290
relates to https://github.com/Homebrew/homebrew-core/pull/64699

erwincoumans, 2020-11-18:
Not all combinations work, I would use the settings from this script:
https://github.com/bulletphysics/bullet3/blob/master/build_cmake_pybullet_double.sh

chenrui333, 2020-12-23:
yeah, it works for me. Thanks @erwincoumans!

closed 2020-12-23 by chenrui333