bulletphysics/bullet3 #3157: URDF Dissociates When using resetBasePositionAndOrientation(). (Noob Question?)

rohit-kumar-j, 2020-11-19:
This is probably a noob question. XD

But my URDF is dissociating as I'm changing the orientation of the base using resetBasePositionAndOrientation([1],[0,0,1],[1,0,0,x]) 
where 'x' is constantly being increased as shown in the terminal below. How can I correct this error? Am I supposed to change the URDF itself? I can't get my head around this. :0

[image]

Ps: I'm very new to Linux and PyBullet in general. 

Here is the URDF...

[V14_URDF.zip](https://github.com/bulletphysics/bullet3/files/5568949/V14_URDF.zip)

erwincoumans, 2020-11-19:
[1,0,0,0] is a unit quaternion, don't make the length longer. If you want to use euler angles, use pybullet.getQuaternionFromEuler

closed 2020-11-19 by erwincoumans

rohit-kumar-j, 2020-11-19:
Thank you for your response!

I figured out how to query using getEulerFromQuaternion() and apply
iteratively it on resetBasePositionAndOrientation(). I was able to reorient
it correctly within 30 mins.
Thank you for making Pybullet opensource!

Warm Regards,
Rohit Kumar J
MIT, Manipal University