bulletphysics/bullet3 #3164: What is getQuaternionSlerp()?

rohit-kumar-j, 2020-11-25:
@erwincoumans In the file: [humanoidMotionCapture.py](https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/humanoidMotionCapture.py) there is a p.getQuaternionSlerp() in line 325. What does it do?

Warm Regards,
Rohit

rohit-kumar-j, 2020-11-29:
Okay, So I figured it out: (I guess..please correct me if I'm wrong...:) )
p.getQuaternionSlerp(<array of intial quaternion values>,<array of final quaternion values>,<change in respective quaternion values>) #all elements are arrays
example:
leftShoulderRot = p.getQuaternionSlerp(leftShoulderRotStart, leftShoulderRotEnd, frameFraction)

erwincoumans, 2020-12-24:
sletp = spherical interpolation

closed 2020-12-24 by erwincoumans

rohit-kumar-j, 2020-12-27:
Thank you @erwincoumans I found a video explaining the SLERP function: [LINK](https://www.youtube.com/watch?v=BXajpAy5-UI)

Is there a way to do a `getEulerSlerp()`? Where can I find the definition of `pybullet.getQuaternionSlerp()?`