bulletphysics/bullet3 #3171: Updating to 3.0.7 from 3.0.6 make constraint unstable

Dongwon-Son, 2020-11-29:
I use pybullet to RL and the task is pick-and-place of dish with ur5e. 
It works well in version 3.0.6, but after updating to 3.0.7 it shows unstable behavior when gripping dish, such as chattering of last joint and sudden blow-up of last 3 joints of ur5e.
Also, this situaiton is not found when I drop version to 3.0.6 again.
Could you advise about cause and solution?

additional detail :
dish : mesh / gripper : mesh
control : torque control with inverse dynamics through pybullet.calculateInverseDynamics (from desired acceleration)
solver max iteration : 150
friction of dish and gripper link : 0.7
timestep : 0.030 / substep : 10 through pybullet.setPhysicsEngineParameter => the resultant physics timestep is 3ms

erwincoumans, 2020-11-29:
Can you provide a self-contained reproduction case?

Dongwon-Son, 2020-11-30:
As you advised, not using substeps (pybullet.setPhysicsEngineParameter(substeps=1)) solves the problems.
before : 
timestep: 0.030 / substep : 10 => 3.0.6 : stable / 3.0.7 : unstable
after:
timestep: 0.003 / substep : 1 => 3.0.6 : stable / 3.0.7 : stable
Thanks!!

closed 2020-11-30 by Dongwon-Son