bulletphysics/bullet3 #3175: Pybullet does not install on 10.15.7

themightyoarfish, 2020-12-01:
I'm using Python 3.8.6 in a venv, and `python setup.py install` yields a bunch of errors, starting with
```
examples/ThirdPartyLibs/zlib/gzlib.c:264:18: error: implicit declaration of function 'lseek' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
```

Clang version is 12.0.0

The full output is here (too long for github): https://pastebin.com/yLEE1Gzc

What could this be caused by?

erwincoumans, 2020-12-01:
Where is -Werror defined?

erwincoumans, 2020-12-24:
I tried pip3 install pybullet on various new machines (Mac OSX 10.16 Big Sur etc), no issue. Cannot reproduce, lets close

closed 2020-12-24 by erwincoumans

themightyoarfish, 2021-01-14:
Here' how I can reproduce this problem:

- MacOS Catalina 10.15.7 
- Python 3.8.7 (from homebrew)
- A `requirements.txt` with only `pybullet~=2.8.7` in it

```
python3 -m venv test_venv
source test_venv/bin/activate
pip install -r requirements.txt --no-cache-dir
```

Maybe the problem exists only in this specific verions

erwincoumans, 2021-01-14:
> Here' how I can reproduce this problem:
>
> * MacOS Catalina 10.15.7
> * Python 3.8.7 (from homebrew)
> * A `requirements.txt` with only `pybullet~=2.8.7` in it
>
> ```
> python3 -m venv test_venv
> source [...]

please try using the latest version

themightyoarfish, 2021-01-14:
Yeah it works with the latest one, but I'd be curious what the problem is, as that version used to work. Anyhow, might be useful for others seeing this issue.