bulletphysics/bullet3 #3186: Failed to pip install pybullet==2.6.9

yiigu, 2020-12-05:
In order to achieve the extra requirement for robosuite, I tried to install pybullet 2.6.9, but it failed.

The error message is shown below.

ERROR: Command errored out with exit status 1: /Users/yigu/anaconda3/envs/rb_bench/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/0h/qr0hdnfs601cy71ylh8qss5c0000gn/T/pip-install-un7qf8b0/pybullet_d89e45bc311846e58a432096d9eaf7be/setup.py'"'"'; __file__='"'"'/private/var/folders/0h/qr0hdnfs601cy71ylh8qss5c0000gn/T/pip-install-un7qf8b0/pybullet_d89e45bc311846e58a432096d9eaf7be/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /private/var/folders/0h/qr0hdnfs601cy71ylh8qss5c0000gn/T/pip-record-tu2cabd1/install-record.txt --single-version-externally-managed --user --prefix= --compile --install-headers /Users/yigu/.local/include/python3.7m/pybullet Check the logs for full command output.

The system information is:
- Mac os Big Sur 11.0.1
- conda 4.9.2
- python 3.7.9

Also, I cannot find the relevant files in the private or .local folder. Could anyone help me with that?

JimmyRooster, 2020-12-21:
[Stack overflow on creating pip logs](https://stackoverflow.com/questions/53844927/where-are-pip-installation-logs#answers)

Try updating pip and removing the cache. Can you install other pip packages?

erwincoumans, 2020-12-23:
I recommend trying the latest version, instead of pinning to a very old version.
pip(3) install pybullet --upgrade --user

closed 2020-12-23 by erwincoumans