bulletphysics/bullet3 #3188: Inverse dynamics with floating base model results in error

thvhauwe, 2020-12-07:
In PyBullet 3.0.7, `calculateInverseDynamics` results in the error message:
```b3Printf: b3Warning[examples/SharedMemory/PhysicsClientSharedMemory.cpp,1073]:

b3Printf: Inverse Dynamics computations failed
pybullet.error: Error in calculateInverseDynamics, please check arguments.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "MWE_inv_dynamics.py", line 27, in <module>
    print(bullet.calculateInverseDynamics(id_robot, [0.0]*n, [0.0]*n, [0.0]*n))
SystemError: <built-in function calculateInverseDynamics> returned a result with an error set
```

**only** when the option `useFixedBase` in `loadURDF` is set to `False`

Using this minimal code as an example (basic URDF file w/ 2 revolute joints provided in attachment):

```
import pybullet as bullet
import time

# Parameters:
robot_base = [0., 0., 0.]
robot_orientation = [0., 0., 0., 1.]

# Initialize Bullet Simulator
id_simulator = bullet.connect(bullet.GUI)  # or bullet.DIRECT for non-graphical version

USE_FIXED_BASE = True

if USE_FIXED_BASE:
    print("In fixed Base...")
    id_robot = bullet.loadURDF("test_inv_dyn.urdf", robot_base, robot_orientation, useFixedBase=True)
    n=2
    print(bullet.calculateInverseDynamics(id_robot, [0.0]*n, [0.0]*n, [0.0]*n))
    print("In fixed Base... SUCCES")
    

USE_FIXED_BASE = False

if not USE_FIXED_BASE:
    print("In floating Base...")
    id_robot2 = bullet.loadURDF("test_inv_dyn.urdf", robot_base, robot_orientation, useFixedBase=False)
    n=2+6
    print(bullet.calculateInverseDynamics(id_robot, [0.0]*n, [0.0]*n, [0.0]*n))
    print("In floating Base... SUCCES")
```

Any advice on how to proceed is greatly appreciated. Possibly linked to #541.
[test_inv_dyn.zip](https://github.com/bulletphysics/bullet3/files/5653331/test_inv_dyn.zip)

JinchengWang, 2020-12-22:
You can try using `flags=1` as in https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_utils/pd_controller_stable.py
For your example that would be
```
if not USE_FIXED_BASE:
    print("In floating Base...")
    id_robot2 = bullet.loadURDF("test_inv_dyn.urdf", robot_base, robot_orientation, useFixedBase=False)
    n=2+7
    print(bullet.calculateInverseDynamics(id_robot2, [0.0]*6+[1., 0., 0.], [0.0]*n, [0.0]*n, flags=1))
    print("In floating Base... SUCCES")
```
Note that the 7th coordinate is for the 4th component (in the order of i,j,k,1) of quaternion base rotation.

thvhauwe, 2020-12-22:
Thanks, this seems to resolve the issue! However, it remains unclear to me what behaviour is changed as this keyword `flags` is undocumented. The example uses this undocumented keyword both for `calculateInverseDynamics` and `calculateMassMatrix`. I'll leave this issue open in case someone can add/comment on the `flags` keyword.

cryscan, 2021-02-09:
According to [here](https://github.com/bulletphysics/bullet3/blob/abea1a848411cf53385fb8288c89db05e5751ef7/examples/SharedMemory/PhysicsServerCommandProcessor.cpp#L11132), setting the `flags` to 1 uses RBD Model (I don't know what exactly it is, but it's for stable PD controller) instead of solving inverse dynamics directly. However, setting `flag` to 0 when keeping the dimension of `q` to `dof + 7` and that of `dq`, `ddq` to `dof + 6` may also work.

closed 2021-03-21 by erwincoumans

zzilch, 2023-08-28:
I met similar issue for `calculateJacobian` and `calculateInverseKinematics`.
The documentation of `calculateJacobian` says:  "The function call takes the full description of the kinematic state, this is because
calculateInverseDynamics is actually called first". 
Is this the same problem?