bulletphysics/bullet3 #3191: pybullet addUserDebugText not work

guodashun, 2020-12-08:
Hi, today I just wanted to use `addUserDebugText()` api to add some msg for my program.
But I met a situation similar to the following:
Firstly, I added debug msg before the `reset()` function like this:

```python
obs, reward, done, info = env.step(action)
if done:
    env.p.addUserDebugText(sth) # in the env, p = BulletClient(p.GUI)
    sleep(2)
    obs = env.reset()
```

But the fact was, the program was indeed terminated by `sleep(2)`, whereas there was nothing on the screen.

Then I tried exchanging the `addUserDebugText()`  and `reset()` like this:

```python
obs, reward, done, info = env.step(action)
if done:
    obs = env.reset()
    env.p.addUserDebugText(sth) # in the env, p = BulletClient(p.GUI)
    sleep(2)
```

It just worked, but it really confused me.

The question is why my first piece of code was not working?

erwincoumans, 2020-12-23:
Indeed, resetSimulation will remove objects and debug text.

closed 2020-12-23 by erwincoumans

guodashun, 2021-01-07:
> Indeed, resetSimulation will remove objects and debug text.

But in the first case before `sleep(2)`, there should be sth on the screen. Even I've tried `p.stepSimulation()` before `sleep(2)` but it didn't work yet.