bulletphysics/bullet3 #3194: CCD no motion rise an assert.

AndreaCatania, 2020-12-11:
I'm debugging this Godot issue: https://github.com/godotengine/godot/issues/43868. When CCD is enabled an assert is raised.

Basically, this `normalize` function https://github.com/bulletphysics/bullet3/blob/master/src/BulletCollision/CollisionDispatch/btCollisionWorld.cpp#L1040 is triggering the assert because the given vector is really little: It's little because there was no motion. 

Considering the [below line](https://github.com/bulletphysics/bullet3/blob/master/src/BulletCollision/CollisionDispatch/btCollisionWorld.cpp#L1041) this may be a case not correctly handled, though I don't know for sure.

Is the following change enough or something else should also be done?
```diff
	btSingleSweepCallback(const btConvexShape* castShape, const btTransform& convexFromTrans, const btTransform& convexToTrans, const btCollisionWorld* world, btCollisionWorld::ConvexResultCallback& resultCallback, btScalar allowedPenetration)
		: m_convexFromTrans(convexFromTrans),
		  m_convexToTrans(convexToTrans),
		  m_world(world),
		  m_resultCallback(resultCallback),
		  m_allowedCcdPenetration(allowedPenetration),
		  m_castShape(castShape)
	{
		btVector3 unnormalizedRayDir = (m_convexToTrans.getOrigin() - m_convexFromTrans.getOrigin());
-		btVector3 rayDir = unnormalizedRayDir.normalized();
+		btVector3 rayDir = unnormalizedRayDir.fuzzyZero() ? btVector(0,0,0) : unnormalizedRayDir.normalized();
		///what about division by zero? --> just set rayDirection[i] to INF/BT_LARGE_FLOAT
		m_rayDirectionInverse[0] = rayDir[0] == btScalar(0.0) ? btScalar(BT_LARGE_FLOAT) : btScalar(1.0) / rayDir[0];
		m_rayDirectionInverse[1] = rayDir[1] == btScalar(0.0) ? btScalar(BT_LARGE_FLOAT) : btScalar(1.0) / rayDir[1];
		m_rayDirectionInverse[2] = rayDir[2] == btScalar(0.0) ? btScalar(BT_LARGE_FLOAT) : btScalar(1.0) / rayDir[2];
		m_signs[0] = m_rayDirectionInverse[0] < 0.0;
		m_signs[1] = m_rayDirectionInverse[1] < 0.0;
		m_signs[2] = m_rayDirectionInverse[2] < 0.0;

		m_lambda_max = rayDir.dot(unnormalizedRayDir);
	}
```

erwincoumans, 2020-12-23:
Sorry for the long delay, I haven't had time to look into this yet. Looks good to me. Thanks!

AndreaCatania, 2020-12-23:
Nup, I'll do a pr tomorrow. Thanks!

PR #3799 (2021-04-26): Fixes assertion caused by a 0len vector normalization
fixes #3194

supersede #3798
commit c0efe069 (2021-04-26): Fixes assertion caused by a 0len vector normalization

fixes #3194
closed 2021-04-26 by erwincoumans