bulletphysics/bullet3 #3197: Using Vulkan to speed up simulation in the future?

TheShermanTanker, 2020-12-16:
As of now the repo's readme states: 
"The entire collision detection and rigid body dynamics can be executed on the GPU.

A high-end desktop GPU, such as an AMD Radeon 7970 or NVIDIA GTX 680 or better. We succesfully tested the software under Windows, Linux and Mac OSX. The software currently doesn't work on OpenCL CPU devices. It might run on a laptop GPU but performance will not likely be very good. Note that often an OpenCL drivers fails to compile a kernel. Some unit tests exist to track down the issue, but more work is required to cover all OpenCL kernels."

My question is: In the future, is there a possibility that the Vulkan Compute API would be used for simulation acceleration instead of OpenCL? Vulkan is a low overhead general purpose GPU API that not only provides rendering pipelines, but also allows general purpose code and calculations to be executed on compatible Vulkan devices (IE directly on the CPU, integrated GPU, and dedicated GPU). From my experience Vulkan also does perform better than OpenCL in terms of speed and is much more stable, and I believe it is also supported by more platforms, although I might be wrong

JimmyRooster, 2020-12-21:
OpenCL has higher overhead to launch kernels than gl/vk compute shaders, but they're also more flexible. OpenCL supports a wider range of hardware than vulkan as well. It would be quite a lot of work to convert everything to vulkan but, depending on the number of kernals used, we could potentially get a decent performance boost.

erwincoumans, 2020-12-23:
I focus on the Tiny Differentiable Simulator for differentiability and GPU acceleration, it allows code-generation into CUDA as well.
See https://www.youtube.com/watch?v=oaYyLU72lSA

closed 2020-12-23 by erwincoumans

TheShermanTanker, 2020-12-27:
> OpenCL has higher overhead to launch kernels than GL/Vulkan compute shaders, but they're also more flexible. OpenCL supports a wider range of hardware than vulkan as well. It would be quite a lot of work to convert everything to vulkan but, depending on the number of kernels used, we could potentially get a decent performance boost.

Hey, you never know if you don't try after all :) 

I could fork and give this a shot, on the 0.00001% off chance whatever I pull off works I get to contribute to Bullet!1!1! Xd

EDIT: Also when I said Vulkan I worded it wrongly, what I meant to say was support for Vulkan alongside of OpenCL :P and besides that OpenGL doesn't let you compute general purpose code like Vulkan does, just clearing that up