bulletphysics/bullet3 #3203: Small Object Contact interaction

jiaqixuac, 2020-12-21:
Hi, I'm working on a simulated surgical arm with tiny objects (~milimeters precision required).
I find it's rather tricky for physics simulation.
[image]
[image]
The torus has a radius of 2cm.
See figures before and after closing the gripper. It makes the torus inside the ground instead of grasping it.
Could you advise on how to enable accurate physics interaction, e.g., grasping for these small objects?

https://github.com/bulletphysics/bullet3/issues/1936

xissburg, 2020-12-21:
Perhaps because the contact points are being merged due to the contact breaking threshold not being low enough? 

Here you can see where they get merged if they're too close https://github.com/bulletphysics/bullet3/blob/b1f76960ff630969050f90784b1f254ad6a7ded8/src/BulletCollision/NarrowPhaseCollision/btPersistentManifold.cpp#L192 and here is where it's calculated and passed to the constructor of `btPersistentManifold` https://github.com/bulletphysics/bullet3/blob/b1f76960ff630969050f90784b1f254ad6a7ded8/src/BulletCollision/CollisionDispatch/btCollisionDispatcher.cpp#L74

Perhaps you could try decreasing the value of `gContactBreakingThreshold` here https://github.com/bulletphysics/bullet3/blob/afa4fb54505fd071103b8e2e8793c38fd40f6fb6/src/BulletCollision/NarrowPhaseCollision/btPersistentManifold.cpp#L26

jiaqixuac, 2020-12-26:
> Perhaps because the contact points are being merged due to the contact breaking threshold not being low enough?
>
> Here you can see where they get merged if they're too close [...]

Thank you for your suggestions.
However, it still fails maybe due to collision margin and other jittering effects.
Could it a better way to scale everything up?
I find that there is a long history of discussion for small objects.

closed 2021-04-26 by erwincoumans