bulletphysics/bullet3 #3207: Exception raised while using libs generated by Visual Studio on Windows

Homura2333, 2020-12-23:
I used Visual Studio to generate the necessary libs and configured them in VS2019, but when I ran a test project, it raised the following exception:

`Unhandled exception at 0x00007FF90CFDF8E8 (ntdll.dll) in Bullet3_test.exe: 0xC0000005: Access violation reading location 0xFFFFFFFFFFFFFFFF.`

And the exception ocurred in the function static void *btAllocDefault(size_t size) in btAlignedAllocator.cpp.

The test codes were as followed:

```
#include "BasicExample.h"

#include "./CommonInterfaces/CommonExampleInterface.h"
#include "./CommonInterfaces/CommonGUIHelperInterface.h"
#include "BulletCollision/CollisionDispatch/btCollisionObject.h"
#include "BulletCollision/CollisionShapes/btCollisionShape.h"
#include "BulletDynamics/Dynamics/btDiscreteDynamicsWorld.h"

#include "LinearMath/btTransform.h"
#include "LinearMath/btHashMap.h"

int main(int argc, char* argv[])
{
	DummyGUIHelper noGfx;

	CommonExampleOptions options(&noGfx);
	CommonExampleInterface* example = BasicExampleCreateFunc(options);

	example->initPhysics();
	example->stepSimulation(1.f / 60.f);
	example->exitPhysics();

	delete example;

	return 0;
}

```

Can anyone help? Thanks very much!

erwincoumans, 2020-12-23:
Do the original projects also crash (BulletExampleBrowser etc)?
Otherwise, try using cmake, and follow those steps: https://github.com/erwincoumans/hello_bullet_cmake

Homura2333, 2020-12-23:
> Do the original projects also crash (BulletExampleBrowser etc)?
> If you build the library using double precision (BT_USE_DOUBLE_PRECISION), your project also needs this definition.
> Try using cmake, and follow those steps: https://github.com/erwincoumans/hello_bullet_cmake

Thank you, it works!

closed 2020-12-23 by Homura2333