bulletphysics/bullet3 #3209: Why are these actions performed together?

Magikod, 2020-12-24:
I loaded a panda robot and want to open the gripper, move to target position, close the gripper. I use time.sleep() to delay the execution time between actions, but it doesn't work. It took a day and still hasn't solved the problem.

Hear is the code;
import pybullet as p
pandaId = p.loadURDF("franka_panda/panda.urdf")

robotEndPos = [1,1,1]
robotEndOrientation = p.getQuaternionFromEuler([0, 3.2 ,0])
targetPositionsJoints = p.calculateInverseKinematics(pandaId, 11, robotEndPos, robotEndOrientation)

finger_target = 0.2
finger_target2 = 0

#open the gripper
for i in range [9,10]:
    p.setJointMotorControl2(pandaId, i, p.POSITION_CONTROL, finger_target,force = 10)
time.sleep(2)

#move to target position
for j in range (7):
    p.setJointMotorControl2(pandaId, j, p.POSITION_CONTROL, targetPositionsJoints[j])
time.sleep(2)

#close the gripper
for k in range [9,10]:
    p.setJointMotorControl2(pandaId, k, p.POSITION_CONTROL, finger_target2, force = 10)

rohit-kumar-j, 2021-01-09:
Try 
```
sleep_time = 100
time.sleep(sleep_time)
```

rohit-kumar-j, 2021-01-09:
Well.. this is not exactly correct (I think), but try this...  (if it doesn't work, try force = 0)

Try: 

```
import pybullet as p
pandaId = p.loadURDF("franka_panda/panda.urdf")

#Open the gripper
target_vel_1 = (finger_target2 - finger_target1) / t1
    p.setJointMotorControl2(pandaId, i, p.VELOCITY_CONTROL, targetVelocity = target_vel_1,force = 10)
time.sleep(t1)

#Move the robot 
for j in range(p.getNumJoints(pandaId))
    s2 = targetPositionsJoints[ j ] # To get final position data 
    s1 = getJointState[ j ] # To get intial position data
    vel = (s2[<final_joint_pos_index>] - s1[<inital_joint_pos_index>]  - ) / t_motion
    p.setJointMotorControl2(pandaId, i, p.VELOCITY_CONTROL, targetVelocity = vel ,force = 10)
time.sleep( t_motion )

#Close the gripper
target_vel_2 = (finger_target1 - finger_target2) / t2
    p.setJointMotorControl2(pandaId, i, p.VELOCITY_CONTROL, targetVelocity = target_vel_2,force = 10)
time.sleep(t2)
```

Magikod, 2021-01-14:
@rohit-kumar-j 
Thanks for your answer.
I solved it.

rohit-kumar-j, 2021-01-14:
Cool! Did my suggestion work? (I'm experimenting this myself and needed to know if it worked)

closed 2021-04-26 by erwincoumans