bulletphysics/bullet3 #3214: Robot stucks into the plane.urdf

KH-HSU, 2021-01-04:
Happy New Year @erwincoumans and everyone

Problem:
When the robot walked from the board, sometimes the robot will stuck into the plane which are shown in pictures below.

If the robot just walked on the plane without the red board in the pictures, the robot will sink a little bit but will not stuck into the plane. ( The red line is for RL experiment without collision, so it will not effect collision.)
 
[image]
[image]

What I have done
1. set timestep to 0.001
2. set numSolverIterations to 1000
3. Use the smallest resolution to generate mesh of the robot in Solidworks (also tried in STL and obj format)

but the problems still exists
Does anyone have met this problem before? Please give me some suggestions, thank you!

KH-HSU, 2021-01-05:
I solved it by 
1. putting the board base under the plane ( I put it just right on the plane before ), 
2. set the numSolverIterations to 1000. 
I'm still curious if there are other solutions because I want to use Heightfield to do experiment.

BluePrintRandom, 2021-01-05:
a plane has zero thickness - try a very short cube instead using the cube
bound.

KH-HSU, 2021-01-05:
@BluePrintRandom 
I'm a bit confused because the plane.urdf given by Pybullet has thickness which can be found in 
[plane.urdf](https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_data/plane.urdf#L22)

I made the red cube by loading [cube.obj](https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_data/cube.obj) 
into urdf

BluePrintRandom, 2021-01-05:
you can use CCD to ensure that a object does not pop through another - but
it's typically easier to simulate a more realistic environment (ie a
thicker floor)

KH-HSU, 2021-01-05:
@BluePrintRandom thank you, I will try CCD.
But the problem still exists when the floor thickness is 4 cm.

I put the green cube under the plane and fix it which is shown in the picture below[image] the red line without collision property is just right on the plane.

I wonder if I just put the floor right on the plane, there might be some small space that will make the robot stuck in the mesh.

closed 2021-04-26 by erwincoumans