bulletphysics/bullet3 #3228: Memory leak in pybullet.calculateInverseKinematics2()

jcoholich, 2021-01-15:
I've been using pybullet.calculateInverseKinematics2() in an RL environment that I wrote, but a memory leak in the function causes my workstation to eventually crash. For now, I've switched to looping through pybullet.calculateInverseKinematics() for IK with multiple end effectors, but I believe this is slower.

According to [Issue #2164](https://github.com/bulletphysics/bullet3/issues/2164), pybullet.calculateInverseKinematics() also had a memory leak that was fixed. Perhaps the problem and solution will be similar.  

This attachment contains a short script to replicate the issue. [test_memory_leak.zip](https://github.com/bulletphysics/bullet3/files/5821840/test_memory_leak.zip)

Memory usage of the program can be monitored using memory-profiler. 
`mprof run python test_memory_leak.py`
`mprof plot`

I am running on Ubuntu 20.04, using pybullet version 3.0.8.

erwincoumans, 2021-01-15:
Thanks for the report, fixed in a PR.

jcoholich, 2021-01-17:
Awesome, thanks!

closed 2021-01-17 by jcoholich