bulletphysics/bullet3 #3236: Jacobian Test in PyBullet

junhyeokahn, 2021-01-20:
Hello, I am testing a ```calculateJacobian``` with floating based robot (```mjcf/humanoid_symmetric.xml```) and have a quick question.
Basically, I am comparing the velocity of the ```left_foot``` achieved from ```getLinkState``` to the velocity I am computing from ```np.dot(J, qdot)``` and having trouble with making them identical.
For ```J```, I got it from ```calculateJacobian```, and for ```qdot```, I stacked base link angular and linear velocity achieved by ```getBaseVelocity``` on the head of joint velocities.
I might wrongly understand the jacobian quantity it gives, but any helps to make these comparison work would be really appreciated.
I am pasting a code I used for the comparison.
Thanks in advance.

```
import time
import os
import sys
cwd = os.getcwd()
sys.path.append(cwd)
import time
import math
import copy
from collections import OrderedDict

import pybullet as p
import pybullet_data
import numpy as np

def get_robot_config(robot):
    nq, nv, na, joint_id, link_id = 0, 0, 0, OrderedDict(), OrderedDict()
    link_id[(p.getBodyInfo(robot)[0]).decode("utf-8")] = -1
    for i in range(p.getNumJoints(robot)):
        info = p.getJointInfo(robot, i)
        if info[2] != p.JOINT_FIXED:
            joint_id[info[1].decode("utf-8")] = info[0]
        link_id[info[12].decode("utf-8")] = info[0]
        nq = max(nq, info[3])
        nv = max(nv, info[4])
    nq += 1
    nv += 1
    na = len(joint_id)

    return nq, nv, na, joint_id, link_id

def set_joint_friction(robot, joint_id, max_force=0):
    p.setJointMotorControlArray(robot, [*joint_id.values()],
                                p.VELOCITY_CONTROL,
                                forces=[max_force] * len(joint_id))

def get_sensor_data(robot, joint_id, link_id):

    sensor_data = OrderedDict()

    base_pos, base_quat = p.getBasePositionAndOrientation(robot)
    sensor_data['base_pos'] = np.asarray(base_pos)
    sensor_data['base_quat'] = np.asarray(base_quat)

    base_lin_vel, base_ang_vel = p.getBaseVelocity(robot)
    sensor_data['base_lin_vel'] = np.asarray(base_lin_vel)
    sensor_data['base_ang_vel'] = np.asarray(base_ang_vel)

    sensor_data['joint_pos'] = OrderedDict()
    sensor_data['joint_vel'] = OrderedDict()
    for k, v in joint_id.items():
        js = p.getJointState(robot, v)
        sensor_data['joint_pos'][k] = js[0]
        sensor_data['joint_vel'][k] = js[1]

    return sensor_data

def get_qdot(robot, sensor_data):
    ## Angular first, Linear Later
    jvel = list(sensor_data['joint_vel'].values())
    ret = np.zeros(6 + len(jvel))
    ret[3:6] = np.copy(sensor_data['base_lin_vel'])
    ret[0:3] = np.copy(sensor_data['base_ang_vel'])
    ret[6:] = np.array(jvel)

    return ret

def get_jacobian(robot, link_idx, sensor_data):
    link_state = p.getLinkState(robot,
                                link_idx,
                                computeLinkVelocity=1,
                                computeForwardKinematics=1)
    jpos = list(sensor_data['joint_pos'].values())
    jvel = list(sensor_data['joint_vel'].values())
    zeros = [0.] * len(jvel)
    jac = p.calculateJacobian(robot, link_idx, link_state[2], jpos, zeros,
                              zeros)
    nv = len(jac[0][0])
    ret = np.zeros((6, nv))
    for row in range(3):
        for col in range(nv):
            ret[row, col] = jac[1][row][col]  # angular
            ret[row + 3, col] = jac[0][row][col]  # linear

    return ret

def get_spatial_velocities(robot, link_idx):
    link_state = p.getLinkState(robot,
                                link_idx,
                                computeLinkVelocity=1,
                                computeForwardKinematics=1)
    ret = np.zeros(6)
    for i in range(3):
        ret[i] = link_state[7][i]  # ang vel
        ret[i + 3] = link_state[6][i]  # lin vel

    return ret

if __name__ == "__main__":

    # Environment Setup
    p.connect(p.GUI)
    p.setAdditionalSearchPath(pybullet_data.getDataPath())
    p.resetDebugVisualizerCamera(cameraDistance=1.5,
                                 cameraYaw=120,
                                 cameraPitch=-30,
                                 cameraTargetPosition=[1, 0.5, 1.5])
    p.setGravity(0, 0, -9.81)
    p.setPhysicsEngineParameter(fixedTimeStep=0.01, numSubSteps=1)

    # Create Robot, Ground
    p.configureDebugVisualizer(p.COV_ENABLE_RENDERING, 0)
    robot = p.loadMJCF("mjcf/humanoid_symmetric.xml")[0]

    p.loadURDF("plane.urdf")
    p.configureDebugVisualizer(p.COV_ENABLE_RENDERING, 1)

    # Robot Configuration
    nq, nv, na, joint_id, link_id = get_robot_config(robot)

    # Joint Friction
    set_joint_friction(robot, joint_id, 2)

    # Run Sim
    t = 0
    dt = 0.01
    count = 0

    while (1):

        # Get SensorData
        sensor_data = get_sensor_data(robot, joint_id, link_id)

        # Jacobian Test
        jac = get_jacobian(robot, link_id['left_foot'], sensor_data)
        vel = get_spatial_velocities(robot, link_id['left_foot'])
        qdot = get_qdot(robot, sensor_data)
        print("-" * 80)
        print("vel")
        print(vel)
        print("jac times qdot")
        print(np.dot(jac, qdot))

        p.stepSimulation()

        time.sleep(dt)
        t += dt
        count += 1
```

junhyeokahn, 2021-01-20:
I just realize the first 6 ```qdot``` of the floating joints should be represented in the base frame.
Also, the jacobian achieved from ```calculateJacobian``` is also represented in the base frame, so I should pre-multiply the base rotation matrix.
I will close this issue.

closed 2021-01-20 by junhyeokahn

YunxiTang, 2022-01-13:
> I just realize the first 6 `qdot` of the floating joints should be represented in the base frame. Also, the jacobian achieved from `calculateJacobian` is also represented in the base frame, so I should pre-multiply the base rotation matrix. I will close this issue.

Hello,
Thanks for your code sharing firstly and I also try to verify the Jacobian of a floating base system following your example. I found that the angular velocity matches the result of getLinkstate() but the linear velocity is slightly different from (the CoM offset of link is considered). Any help will be appreciated! Thanks.