bulletphysics/bullet3 #3242: resetJointState unexpected behavior

lukashermann, 2021-01-22:
Hi,
I am using pybullet to simulate a franka emika panda robot. The environemt uses the basic gym env structure with a reset and a step function. In the reset function the robot is reset to a desired joint configuration by calling resetJointState(...) and in the step function the robot is controlled via IK and setJointMotorControl2.
I recorded some trajectories and tried to replay them sequentially by calling reset with the recorded joint states and step with the recorded actions. There I noticed a weird behavior: resetJointState doesn't actually reset the joints to the desired position (or rather, it seems like they are reset but in an instant swing back to the previous position). This does not happen when only calling the reset function without having called setJointMotorControl2 before.

I check some pybullet examples and found that for example in 
https://github.com/bulletphysics/bullet3/blob/afa4fb54505fd071103b8e2e8793c38fd40f6fb6/examples/pybullet/gym/pybullet_envs/bullet/kuka.py#L57
resetJointState is called together with setJointMotorControl2 (with the exact same joint positions). Shouldn't resetJointState FULLY reset the simulation? Following this example, I tried using both, and this fixed my problem. Is this known and wanted behavior?

mees, 2021-01-22:
I think this is a expected behaviour as without setJointMotorControl2 the controller stills will try to go to the old position.

erwincoumans, 2021-01-22:
Yes, the 'reset' functions just set the current joint pose/velocity, and are not meant to be used during the simulation, only when setting up the simulation, or during a real 'reset' (after calling resetSimulation). 
Since setJointMotorControl2 is still active, it will drive the joint back to the desired position/velocity.

closed 2021-01-22 by erwincoumans