bulletphysics/bullet3 #3248: saveState memory leak.

lewisboyd, 2021-01-28:
Hi,

I think I found a bug in PyBullet. The issue is that calling p.saveState() seems to take up memory that is never released even if the reference to the state is manually deleted using pythons del function. This is an issue for me because I would like to use these functions in my reset method for a gym environment but this makes it unusable for training since it eventually leads to an out of memory error.

The bug can be recreated by repeatedly calling p.saveState() and watching the memory usage for the python process in the resource monitor. 

Tested on a windows 10 machine using pybullet version 3.0.7.

erwincoumans, 2021-01-28:
pybullet.saveState saves the state to memory, you are supposed to call it only once, and then use loadState from then on. Alternatively, use saveState to disk, instead of memory.

closed 2021-01-28 by erwincoumans

lewisboyd, 2021-01-29:
Thanks for getting back to me so quickly! Apologies for misunderstanding the usage of saveState!