bulletphysics/bullet3 #3258: LICENSE of files

str4si, 2021-02-08:
In the LICENSE.txt in the root directory it states "The files in this repository are licensed under the zlib license, except for the files under 'Extras' and examples/ThirdPartyLibs."

but under  'Extras'  there are no other license files in the subdirectories so it is unclear under which license they fall.

and in examples/ThirdPartyLibs only a subset of the directories has license files, the situation of the other files and directories is equally unclear.

this prevents anyone taking software licenses serious from using bullet3, could this situation be improved ?

erwincoumans, 2021-02-08:
Dis you check the license header inside any of the files? Which specific file(s) have an unclear license?

str4si, 2021-02-08:
ok, thanks. i randomly opened https://github.com/bulletphysics/bullet3/blob/master/Extras/BulletRobotics/premake4.lua which does not have license information, but on the other hand it is a build file. checking some .cpp and .h files i found no missing license information.

erwincoumans, 2021-02-08:
I added license files for all directories in Extras and examples/ThirdPartyLibs.
Often the license information is at the top of source code files, the build files in Bullet as also zlib license, if you want to re-use them.

closed 2021-02-08 by erwincoumans