bulletphysics/bullet3 #3259: How ${BULLET_PHYSICS_SOURCE_DIR} is defined?

yangli-flexiv, 2021-02-08:
I am using Ubuntu 18.04, and have run the build command:
```
./build_cmake_pybullet_double.sh
./premake4_linux64 --double gmake
cd gmake
make
```
When I tried to `cd` to `HelloWorld` example, and built the code:
```
mkdir build && cd build
cmake .. -DCMAKE_INSTALL_PREFIX=~/install -DBUILD_THIRDPARTY=ON
make
```
and I got error, saying that 
```
fatal error: btBulletDynamicsCommon.h: No such file or directory
 #include "btBulletDynamicsCommon.h"
```
I checked the CMakeLists.txt file, and by printing out the `INCLUDE_DIRECTORIES: ${BULLET_PHYSICS_SOURCE_DIR}/src`, and I found that `${BULLET_PHYSICS_SOURCE_DIR}` is not defined. 

I don't know whether I need to run some other command to configure `${BULLET_PHYSICS_SOURCE_DIR}`? 

Thank you a lot!

erwincoumans, 2021-02-08:
I think cmake defines that variable, if ran from the root directory.

./build_cmake_pybullet_double.sh
This should already build all the examples, you shouldn't need to build them again.

We use this to build Bullet from a different repo:

~~~~~~~~~~~~~~~~~
cmake  -DUSE_DOUBLE_PRECISION=ON -DCMAKE_DEBUG_POSTFIX="" -DINSTALL_LIBS=ON -DCMAKE_BUILD_TYPE=Release  -DCMAKE_INSTALL_PREFIX:PATH=local_install  ..
make -j4
make install
~~~~~~~~~~~~~~~~~

See also this simple cmake example:
https://github.com/erwincoumans/hello_bullet_cmake

yangli-flexiv, 2021-02-08:
Hi @erwincoumans,

Thank you very much for your reply. I am not sure what you mean by `root` directory, this command `build_cmake_pybullet_double.sh` is located in the directory of `bulletphysics`. 

Did you mean that when I ran `./build_cmake_pybullet_double.sh`, I don't need to `cd` to each example and built them separately?

Thanks!

closed 2021-04-26 by erwincoumans