bulletphysics/bullet3 #3261: error in installation of pybullet in python 3.6(pip install pybullet)

khushal143, 2021-02-09:
I used different commands but installation failed
pip install pybullet==2.1.9
pip install -U -no-cache-dir pybullet
I tested both ways to build v2.1.9 and still get the same error message.
[image]
   running build_ext
    building 'pybullet' extension
    error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/
    ----------------------------------------
ERROR: Command errored out with exit status 1: 'C:\Users\khushal khan\Anaconda3\envs\universe\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\KHUSHA~1\\AppData\\Local\\Temp\\pip-install-53gdy2o4\\pybullet_8bd5f4d8ef3244a19fda3292aa305928\\setup.py'"'"'; __file__='"'"'C:\\Users\\KHUSHA~1\\AppData\\Local\\Temp\\pip-install-53gdy2o4\\pybullet_8bd5f4d8ef3244a19fda3292aa305928\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\KHUSHA~1\AppData\Local\Temp\pip-record-0mq9efdp\install-record.txt' --single-version-externally-managed --compile --install-headers 'C:\Users\khushal khan\Anaconda3\envs\universe\Include\pybullet' Check the logs for full command output.

yagi-osaka, 2021-02-19:
I have the same problem.

erwincoumans, 2021-02-19:
As the error reports, you need a full installation of Visual Studio 2019 with C++ support, it is free, download it from Microsoft.

closed 2021-02-19 by erwincoumans