bulletphysics/bullet3 #3271: Limit character movements

tfederico, 2021-02-15:
Hi,

I was wondering if there is a way to lock the lower-body part of the humanoid character.

What I would like to do is being able to move the upper part of the body and block the lower part, without making the humanoid fall when it performs movements with the arms.

I noticed that if I set `useFixedBase` to `True`, the root is locked, but the remaining parts of the body can fully rotate around the root. Is there a way to do something similar with the legs? What would happen if I were to increase the inertia of the legs joints?

Many thanks in advance

tfederico, 2021-02-17:
@erwincoumans if I increase the mass of the lower-body parts, would the center of mass automatically move closer to the ground?

rohit-kumar-j, 2021-02-18:
@tfederico , you can try:
```
pos = <whatever_existing _joint_position_you_want>
maxForce = <It is stated as 500 in the Pybullet quick start guide, but i think you can make it higher>
p.setJointMotorControl2(<robotUId>,<joint_index>,controlMode=p.POSITION_CONTROL,tragetPosition=pos,maxForce=maxForce)
```
Warm  Regards, 
Rohit

erwincoumans, 2021-03-07:
You can edit the URDF file and make the joints "fixed", instead of "revolute" / "continuous" / "spherical"

closed 2021-03-07 by erwincoumans

tfederico, 2021-03-30:
Hi @erwincoumans, sorry for the naive question but wouldn't make the joints "fixed" just block them from rotating? I would actually find a way to anchor the character so that, even if it moves the upper body, it doesn't fall