bulletphysics/bullet3 #3273: OpenGL GUI Doesn't Render Bodies Properly

agupta231, 2021-02-19:
Hi PyBullet team!

I'm working with a team to run RL experiments on quadruped that we designed. We built the model in Solidworks and generated the URDF from that. Once we had imported into PyBullet, we got very weird behavior:

[image]

For whatever reason, it seems as if the GPU/GUI doesn't render some of the bodies right, but as seen in the output from `getCameraImage`, the bodies clearly exist within the simulation. Furthermore, I can control the invisible bodies from code, see the bodies get rendered in `getCameraImage` and see the robot react to the movement (albeit I can't actually see the missing bodies themselves) in the GUI. It's incredibly bizarre cause even the shadows of the bodies aren't being rendered, even though the body exists.

We're tearing out head out on this because one of our experiments is to simulate running code on a real robot, and we need to use `setRealtimeSimulation` for the most realistic scenario. Additionally, with the overhead of the CPU renderer, we would really prefer if we could keep things simple and fast and just use the GUI.

We've ran out stack on other URDFs and we've had success. Our code is a bit involved, so I didn't include it yet for brevity's sake. Please let me know if I can provide any other information to help out. Thank you so much!

erwincoumans, 2021-02-19:
Can't tell without repro case, but most likely your models have too many vertices. Please simply them a lot, for example using Blender (mesh reduction). Sometimes, people even add screws of thousands of vertices that are invisible. We use a very conservative limit on the GPU memory usage.

agupta231, 2021-02-19:
Sounds good! I'll try to simplify the models and I'll let you know if that works! Thank you for the quick response!

agupta231, 2021-02-19:
You're guess was spot on--one of my teammates had exported the STLs with the screw included and the number of vertices was ridiculous. Reduced them via meshlab and everything works great. Thanks again for the help!

closed 2021-02-19 by agupta231