bulletphysics/bullet3 #3280: Please do not use bundled zlib because of SEGVs due to conflicts with pre-installed zlib

yurivict, 2021-02-22:
There is a segmentation fault on FreeBSD 12 STABLE:
```
#0 atomic_load_p (a=0x0, mo=atomic_memory_order_relaxed) at /usr/src/contrib/jemalloc/include/jemalloc/internal/atomic.h:55
#1 rtree_leaf_elm_bits_read (tsdn=<optimized out>, rtree=<optimized out>, elm=0x0, dependent=true) at /usr/src/contrib/jemalloc/include/jemalloc/internal/rtree.h:175
#2 rtree_szind_slab_read (tsdn=0x80c7f7090, rtree=<optimized out>, rtree_ctx=<optimized out>, key=131, dependent=true, r_szind=<optimized out>, r_slab=<optimized out>) at /usr/src/contrib/jemalloc/include/jemalloc/internal/rtree.h:464
#3 ifree (tsd=<optimized out>, ptr=0x83, tcache=0x80c7f7250, slow_path=false) at jemalloc_jemalloc.c:2210
#4 __free (ptr=<optimized out>) at jemalloc_jemalloc.c:2382
#5 0x0000000801e4219f in inflateReset2 () from /usr/local/lib/python3.7/site-packages/pybullet.so
#6 0x00000008017c9af5 in inflateInit2_ (strm=0x7fffdf9f8cf0, windowBits=15, version=<optimized out>, stream_size=<optimized out>) at /usr/src/sys/contrib/zlib/inflate.c:231
#7 inflateInit_ (strm=0x7fffdf9f8cf0, version=<optimized out>, stream_size=<optimized out>) at /usr/src/sys/contrib/zlib/inflate.c:244
#8 0x000000080d155a2b in ?? () from /usr/local/lib/dri/i965_dri.so
#9 0x000000080d22bb79 in ?? () from /usr/local/lib/dri/i965_dri.so
#10 0x000000080d16dcf0 in ?? () from /usr/local/lib/dri/i965_dri.so
#11 0x000000080d082591 in ?? () from /usr/local/lib/dri/i965_dri.so
#12 0x000000080cf7cd60 in ?? () from /usr/local/lib/dri/i965_dri.so
#13 0x0000000801eb5567 in gltLoadShaderPair () from /usr/local/lib/python3.7/site-packages/pybullet.so
#14 0x0000000801ea91d5 in GLInstancingRenderer::InitShaders() () from /usr/local/lib/python3.7/site-packages/pybullet.so
#15 0x0000000801eae6cb in SimpleOpenGL3App::SimpleOpenGL3App(char const*, int, int, bool, int, int, int, int) () from /usr/local/lib/python3.7/site-packages/pybullet.so
#16 0x0000000801e996ae in OpenGLExampleBrowser::init(int, char**) () from /usr/local/lib/python3.7/site-packages/pybullet.so
#17 0x0000000801d2a04c in ExampleBrowserThreadFunc(void*, void*) () from /usr/local/lib/python3.7/site-packages/pybullet.so
#18 0x0000000801e89afb in ?? () from /usr/local/lib/python3.7/site-packages/pybullet.so
#19 0x00000008005f208c in thread_start (curthread=0x802ae6d00) at /usr/src/lib/libthr/thread/thr_create.c:292
#20 0x0000000000000000 in ?? ()

```

erwincoumans, 2021-02-22:
Why? Isn't the bundled zlib statically linked?

yurivict, 2021-02-22:
The user reported that:
```
 the i965_dri module dynamically links the system version of zlib:

/usr/local/lib/dri/i965_dri.so:
        libz.so.6 => /lib/libz.so.6 (0x80086a000)
        libglapi.so.0 => /usr/local/lib/libglapi.so.0 (0x800886000)
        libdrm_intel.so.1 => /usr/local/lib/libdrm_intel.so.1 (0x8008e6000)
        libdrm_radeon.so.1 => /usr/local/lib/libdrm_radeon.so.1 (0x80090e000)
        libdrm.so.2 => /usr/local/lib/libdrm.so.2 (0x80091c000)
        libexpat.so.1 => /usr/local/lib/libexpat.so.1 (0x800933000)
        libthr.so.3 => /lib/libthr.so.3 (0x8007c0000)
        libc++.so.1 => /usr/lib/libc++.so.1 (0x801b31000)
        libcxxrt.so.1 => /lib/libcxxrt.so.1 (0x800960000)
        libm.so.5 => /lib/libm.so.5 (0x800982000)
        libc.so.7 => /lib/libc.so.7 (0x800253000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x8009b4000)
        libpciaccess.so.0 => /usr/local/lib/libpciaccess.so.0 (0x8009ce000)
```
and the conflict arises.

This patch helps:
```
--- setup.py.orig	2021-02-22 06:17:11 UTC
+++ setup.py
@@ -160,21 +160,6 @@ sources = ["examples/pybullet/pybullet.c"]\
 +["examples/ThirdPartyLibs/minizip/ioapi.c"]\
 +["examples/ThirdPartyLibs/minizip/unzip.c"]\
 +["examples/ThirdPartyLibs/minizip/zip.c"]\
-+["examples/ThirdPartyLibs/zlib/adler32.c"]\
-+["examples/ThirdPartyLibs/zlib/compress.c"]\
-+["examples/ThirdPartyLibs/zlib/crc32.c"]\
-+["examples/ThirdPartyLibs/zlib/deflate.c"]\
-+["examples/ThirdPartyLibs/zlib/gzclose.c"]\
-+["examples/ThirdPartyLibs/zlib/gzlib.c"]\
-+["examples/ThirdPartyLibs/zlib/gzread.c"]\
-+["examples/ThirdPartyLibs/zlib/gzwrite.c"]\
-+["examples/ThirdPartyLibs/zlib/infback.c"]\
-+["examples/ThirdPartyLibs/zlib/inffast.c"]\
-+["examples/ThirdPartyLibs/zlib/inflate.c"]\
-+["examples/ThirdPartyLibs/zlib/inftrees.c"]\
-+["examples/ThirdPartyLibs/zlib/trees.c"]\
-+["examples/ThirdPartyLibs/zlib/uncompr.c"]\
-+["examples/ThirdPartyLibs/zlib/zutil.c"]\
 +["examples/Importers/ImportColladaDemo/LoadMeshFromCollada.cpp"]\
 +["examples/Importers/ImportObjDemo/LoadMeshFromObj.cpp"]\
 +["examples/Importers/ImportObjDemo/Wavefront2GLInstanceGraphicsShape.cpp"]\
@@ -438,7 +423,7 @@ elif _platform == "darwin":
   +["examples/OpenGLWindow/MacOpenGLWindowObjC.m"]
 else:
   print("bsd!")
-  libraries = ['GL', 'GLEW', 'pthread']
+  libraries = ['GL', 'GLEW', 'pthread', 'z']
   os.environ['LDFLAGS'] = '-L/usr/X11R6/lib'
   CXX_FLAGS += '-D_BSD '
   CXX_FLAGS += '-I/usr/X11R6/include '
```

erwincoumans, 2021-03-04:
If you care, you can create an option in the setup.py process for system-wide zlib (disabled by default!).
I'd like to keep PyBullet to be self-contained (note that we don't need to link against GL and GLEW btw, it can be removed).
Not all  systems have zlib installed system wide.
Closing for now, so if you care, please create a pull request with setup.py option (not changing the default behavior).

closed 2021-03-04 by erwincoumans