bulletphysics/bullet3 #3284: Simulating a ballbot robot

rshum19, 2021-02-23:
Hello, 

This is more a question about the capabilities of the physics simulator than an issue with it. I'm trying to simulate a "ballbot" type robot. This is a robot that has a cylindrical body that balances on top of a single spherical wheel [[see video]](https://www.youtube.com/watch?v=8BtDuzu2WeI&t=2sl). The two issues that we have faced with other simulators are:

1. Simulating the connection between the body and the ball
2. Actuating the ball 

 I'm wondering if you think it would be within pyBullet capabilities to simulate such a robot?  

Thank you,

erwincoumans, 2021-02-23:
Hi, you could attach the ball to the robot torso using a spherical joint. This spherical joint can be actuated in various ways, for example explicit PD control, implicit (stable) PD control or using a spherical joint motor. See this example how to configure this: https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/humanoidMotionCapture.py
An example of a URDF with spherical joint is here:
https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_data/humanoid/humanoid.urdf

rshum19, 2021-02-23:
Thank you for the prompt response. This is very helpful, the actuation of the spherical joint may be the part I was missing.

rshum19, 2021-02-23:
I was under the impression that URDF did to have the spherical joint type according to http://wiki.ros.org/urdf/XML/joint. Does pyBullet have a special interpreter for this type of joints?

erwincoumans, 2021-02-23:
Yes, we reimplemented the URDF parser and added additional features (and not all URDF tags are used). We use Python to enable actuators. Please just check the example I send in previous post.

closed 2021-02-23 by erwincoumans