bulletphysics/bullet3 #3285: resetSimulation does not free memory when using EGL

ywchao, 2021-02-25:
Following #2523, there still seems to be a memory leak when EGL is enabled, even with resetSimulation. Here is a code snippet to reproduce the issue:

```
import pybullet
import pybullet_utils.bullet_client as bullet_client
import pybullet_data

p = bullet_client.BulletClient(connection_mode=pybullet.DIRECT)

p.setAdditionalSearchPath(pybullet_data.getDataPath())

import pkgutil
egl = pkgutil.get_loader('eglRenderer')
plugin = p.loadPlugin(egl.get_filename(), "_eglRendererPlugin")

for _ in range(1000):
  p.resetSimulation()
  body = p.loadURDF("franka_panda/panda.urdf")
```

This is only using loadURDF as suggested in #2898. With this code I am observing an increasing usage on CPU memory and also GPU memory from using EGL.

RchalYang, 2021-03-04:
Got the same problem

erwincoumans, 2021-03-04:
Thanks for the report. 
You can avoid this by not loading a URDF file each time a simulation is reset. You can get deterministic behavior using saveState/restoreState, and it is much better for reset performance. See here how this is done in some PyBullet Gym environments:
https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_envs/gym_pendulum_envs.py#L20

It is due to a texture not released (colors.png). Fixed here: https://github.com/bulletphysics/bullet3/pull/3296

erwincoumans, 2021-03-04:
The leak should be resolved in version PyBullet 3.09, https://pypi.org/project/pybullet

closed 2021-03-04 by erwincoumans

RchalYang, 2021-03-07:
Thanks for the new version, but it seems it's still happening for some specific URDF file.

I tested it on my machine. It works fine if i load rd2d.urdf. But the GPU Memory consumption keep increase when i load a1/a1.urdf.