bulletphysics/bullet3 #3286: Mutliprocessing and Multiple Pybullet clients

mauicv, 2021-02-27:
Hey,
I'm trying to run as many simulations of an environment in as many processes as possible. As I understand it the following worker is run as a new process where it then creates a new pybullet instance and then within this process adds a number of direct connections and runs them all together once the _EXPLORE message is passed.

```py
import random
from multiprocessing import Process, Pipe
from time import time

_RESET = 1
_CLOSE = 2
_EXPLORE = 3

def Worker(rank, num_processes, childPipe):
    import pybullet_utils.bullet_client as bc
    import pybullet_data
    import pybullet_envs  # noqa
    import pybullet as p

    sims = []
    sims_per_worker = 25

    while True:
        try:
            if not childPipe.poll(0.0001):
                continue
            message = childPipe.recv()
        except (EOFError, KeyboardInterrupt):
            break
        if message == _RESET:
            for i in range(sims_per_worker):
                client = bc.BulletClient(connection_mode=p.DIRECT)
                client.setAdditionalSearchPath(pybullet_data.getDataPath())
                client.setGravity(0, 0, -10)
                client.loadURDF("plane.urdf")
                cubeStartPos = [0, 0, 1]
                cubeStartOrientation = client.getQuaternionFromEuler([
                    random.random(), random.random(), random.random()])
                boxId = client \
                    .loadURDF("r2d2.urdf", cubeStartPos, cubeStartOrientation)
                sims.append(client)
            childPipe.send('Resetting')

        if message == _EXPLORE:
            for i in range(1000):
                for sim in sims:
                    sim.stepSimulation()
            childPipe.send('Finished exploring')
        if message == _CLOSE:
            childPipe.send("close ok")
            break
    childPipe.close()
```

Am I right in assuming this is whats happening? 

My (probably faulty) understanding from [this issue](https://github.com/bulletphysics/bullet3/issues/1925) is that you can run multiple direct environments in one pybullet instance. In that case is the instance unique per process? Suppose I run 4 of these workers as defined above, does that mean I have four pybullet instances, one for each process and then 25 simulations are ran in each? So 100 simulations in total. 

Does it makes sense to do the above? Or have I misunderstood something?

Thanks

erwincoumans, 2021-03-04:
Yes, pybullet_utils.bullet_client allows you to run multiple instances in parallel. The link you shared explains how to do this. Yes indeed, 4 worker processes, each one running 25 sims_per_worker.

closed 2021-03-04 by erwincoumans

mauicv, 2021-03-04:
Thank you, much appreciated!