bulletphysics/bullet3 #3290: Could segmentation mask be used in training?

sodarin, 2021-03-02:
I am learning reinforcement learning and conducting an experiment in KukaDiverseObjectEnv. I wonder whether  I can use the image of segmentation mask instead of the rgb image as the input of neural networks.

erwincoumans, 2021-03-04:
You could, just change the source code, it it Python.

closed 2021-03-04 by erwincoumans

BlGene, 2021-03-04:
eg. with the code below, `obj_arr` is what you want.
```
        if info:
            flag = self.p.ER_SEGMENTATION_MASK_OBJECT_AND_LINKINDEX
        else:
            flag = self.p.ER_NO_SEGMENTATION_MASK

        camera_images = self.p.getCameraImage(width=zoom_width,
                                              height=zoom_height,
                                              viewMatrix=view_mat,
                                              projectionMatrix=self.proj_matrix,
                                              shadow=1, 
                                              lightDirection=light_direction,
                                              renderer=self.p.ER_BULLET_HARDWARE_OPENGL,
                                              flags=flag,
                                              physicsClientId=self.cid)

        hight, width, img_arr, depth_arr, obj_arr =  camera_images
        ```