bulletphysics/bullet3 #3294: BoxShape dimensions sent in the ctor are sometimes not set properly unless the body is re-added

ChronicallySerious, 2021-03-03:
Bullet version: 3.08
BoxShape integration code: https://github.com/sdslabs/Rootex/blob/master/rootex/framework/components/physics/box_collider_component.cpp#L21

In a scene such as this (floor is using BoxShape, sphere is using SphereShape):
[image]

The sphere is expected to fall, which it does but it is not expected to stop mid-air as if the box collider had double the height than it already has. This mid-air situation is screen-capped above.

After the ball has stabilized mid-air, and then if I just change the dimensions of the floor's box collider even a little bit, which btw runs this code:

```cpp
// BoxColliderComponent inherits from PhysicsColliderComponent
void BoxColliderComponent::setDimensions(const Vector3& dimensions)
{
	m_Dimensions = dimensions;
	m_CollisionShape.reset(new btBoxShape(VecTobtVector3(dimensions)));
	m_BoxShape = (btBoxShape*)m_CollisionShape.get();
	setupData();
}

bool PhysicsColliderComponent::setupData()
{
	if (m_Body)
	{
		PhysicsSystem::GetSingleton()->removeRigidBody(m_Body.get());
	}

	if (m_IsMoveable)
	{
		m_Mass = m_Volume * PhysicsSystem::GetSingleton()->getMaterialData(m_Material).specificGravity;
	}
	else
	{
		m_Mass = 0.0f;
	}

	m_CollisionShape->calculateLocalInertia(m_Mass, m_LocalInertia);

	btRigidBody::btRigidBodyConstructionInfo rbInfo(m_Mass, this, m_CollisionShape.get(), m_LocalInertia);
	{
		const PhysicsMaterialData& materialData = PhysicsSystem::GetSingleton()->getMaterialData(m_Material);
		rbInfo.m_restitution = materialData.restitution;
		rbInfo.m_friction = materialData.friction;
	}
	m_Body.reset(new btRigidBody(rbInfo));

	setGravity(m_Gravity);
	setMoveable(m_IsMoveable);
	setKinematic(m_IsKinematic);
	setAngularFactor(m_AngularFactor);
	setSleepable(m_IsSleepable);
	setCCD(m_IsCCD);
	m_Body->setUserPointer(this);

	PhysicsSystem::GetSingleton()->addRigidBody(m_Body.get(), m_CollisionGroup, m_CollisionMask);
	return true;
}
```
...then the sphere falls down to the actual surface of the floor.

Let me know if there is any more specific info I need to provide

ChronicallySerious, 2021-03-03:
Additionally, copying from the above issue ^

>The sphere gets stuck in mid-air (much above the floor) and only falls to the floor when either:
>
>* Floor is set to kinematic and then back
>* Sphere is set to kinematic and then back
>* Floor dimensions are changed

erwincoumans, 2021-03-04:
The problem is most likely in the calling code, not inside the Bullet SDK. Closing for now. Please create a new issue, if you can re-create it in any of the example of the ExampleBrowser or in PyBullet.

closed 2021-03-04 by erwincoumans