bulletphysics/bullet3 #3304: "ResetSimulation does not free memory when using EGL" Still Exists with Pybullet 3.0.9

RchalYang, 2021-03-07:
I recently found that resetSimulation did not free memory when using EGL in some case.

I did notice that in #3285 it says it's fixed in pybullet 3.0.9. It seems the issue still exists. At least when load some specific URDF.

I tested on my machine with the following code adapted from Pybullet examples.

It works fine when I load "r2d2.urdf" but it's not when I'm working with "a1/a1.urdf".

```
import pybullet as p
import time
import pkgutil
egl = pkgutil.get_loader('eglRenderer')
import pybullet_data

p.connect(p.DIRECT)
p.setAdditionalSearchPath(pybullet_data.getDataPath())
plugin = p.loadPlugin(egl.get_filename(), "_eglRendererPlugin")
print("plugin=", plugin)

p.configureDebugVisualizer(p.COV_ENABLE_RENDERING, 0)
p.configureDebugVisualizer(p.COV_ENABLE_GUI, 0)

p.setGravity(0, 0, -10)
a1 = p.loadURDF("a1/a1.urdf")

pixelWidth = 320
pixelHeight = 220
camTargetPos = [0, 0, 0]
camDistance = 4
pitch = -10.0
roll = 0
upAxisIndex = 2

c = 0

while (p.isConnected()):

    p.resetSimulation()

    # idp = p.loadURDF("plane.urdf", [0, 0, -1])
    a1 = p.loadURDF("a1/a1.urdf")

    
    for yaw in range(0, 360, 10):
        start = time.time()
        p.stepSimulation()
        stop = time.time()
        print("stepSimulation %f" % (stop - start))

        #viewMatrix = [1.0, 0.0, -0.0, 0.0, -0.0, 0.1736481785774231, -0.9848078489303589, 0.0, 0.0, 0.9848078489303589, 0.1736481785774231, 0.0, -0.0, -5.960464477539063e-08, -4.0, 1.0]
        viewMatrix = p.computeViewMatrixFromYawPitchRoll(camTargetPos, camDistance, yaw, pitch, roll,
                                                        upAxisIndex)
        projectionMatrix = [
            1.0825318098068237, 0.0, 0.0, 0.0, 0.0, 1.732050895690918, 0.0, 0.0, 0.0, 0.0,
            -1.0002000331878662, -1.0, 0.0, 0.0, -0.020002000033855438, 0.0
        ]

        start = time.time()
        img_arr = p.getCameraImage(pixelWidth,
                                pixelHeight,
                                viewMatrix=viewMatrix,
                                projectionMatrix=projectionMatrix,
                                shadow=1,
                                lightDirection=[1, 1, 1])
        stop = time.time()
        print("renderImage %f" % (stop - start))

p.unloadPlugin(plugin)

```

I'm using pybullet 3.0.9 on GTX 1080

erwincoumans, 2021-03-07:
I checked it, and there is no memory leak in above script, using PyBullet 3.0.9. 

Also, don't load urdf files in a reset function, use saveState/restoreState instead.

closed 2021-03-07 by erwincoumans

ac-93, 2021-03-19:
This script leaks for me too, GPU memory keeps rising when loading EGL plugin but seems fine without. 

Ubuntu 18.04
Nvidia 2080Ti
pybullet 3.1.0