bulletphysics/bullet3 #3305: ImportError: numpy.core.multiarray failed to import

linqingfan, 2021-03-07:
Hi,
When I do import pybullet, it shows the error:
ImportError: numpy.core.multiarray failed to import

May I know what could be the problem?

closed 2021-03-07 by linqingfan

lorepieri8, 2021-03-08:
@erwincoumans  
The Error arise in the pip version 3.0.9, while it's not there in 3.0.8.

This works without errors: 
`pip install pybullet==3.0.8
import pybullet as p`

erwincoumans, 2021-03-08:
What OS and how was Python installed?

lorepieri8, 2021-03-08:
Colab.

Create a new colab and run 

```
!pip install pybullet
import pybullet
```

to reproduce the error.

erwincoumans, 2021-03-08:
pybullet 3.0.9 is compiled against latest numpy (1.20.1). It seems colab still uses 1.19.5
I'll upload a new package (3.0.10) against numpy 1.19.5. Alternatively, pip install numpy==1.20.1

erwincoumans, 2021-03-08:
Should be fixed now with pybullet-3.1.0 compiled against numpy==1.19.5
~~~
!pip install pybullet --upgrade
import pybullet as p
~~~