bulletphysics/bullet3 #3306: import error

linqingfan, 2021-03-07:
Hi,
After I:
sudo pip3 install pybullet

The following error occurs:

Python 3.8.5 (default, Jan 27 2021, 15:41:15) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pybullet
pybullet build time: Mar  4 2021 20:17:30
RuntimeError: module compiled against API version 0xe but this version of numpy is 0xd
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: numpy.core.multiarray failed to import

My numpy is working fine. What could have been the problem?

linqingfan, 2021-03-07:
I solved it by compiling from src

closed 2021-03-07 by linqingfan

erwincoumans, 2021-03-08:
pybullet 3.0.9 is compiled against latest numpy (1.20.1). It seems colab still uses 1.19.5
I'll upload a new package (3.0.10) against numpy 1.19.5. Alternatively, pip install numpy==1.20.1

erwincoumans, 2021-03-08:
Should be fixed now with pybullet-3.1.0 compiled against numpy==1.19.5
~~~
!pip install pybullet --upgrade
import pybullet as p
~~~

Marc-Morcos, 2024-11-13:
Hey, this is an issue again with numpy>2.0

Pboiiiii, 2025-03-01:
> pybullet 3.0.9 is compiled against latest numpy (1.20.1). It seems colab still uses 1.19.5 I'll upload a new package (3.0.10) against numpy 1.19.5. Alternatively, pip install numpy==1.20.1

Got the same issue with pybullet and using numpy 1.20.1 solved the problem.