bulletphysics/bullet3 #3307: Jacobians during simulation.step()

devesh1508, 2021-03-10:
We are trying to understand the implementation a bit in detail. Could you please help me understand the following questions
1. We understand the Jacobian is a function of multi-body states. When the bodies are not in contact, are the Jacobians still computed during simulation.step? We found that sometimes the Jacobians are all zeros? Are the Jacobians when contact is incipient between some bodies in the environment or are they always computed, i.e., even in the case when the bodies are far apart?
2. Is the MLCP problem solved in every step irrespective of whether the bodies are in contact or not. In cases, when the bodies are not in contact, that would result in zero contact forces. Or these are solved only when the bodies are in contact and is decided based on some distance threshold between bodies. This question is also related to the previous question.

Thanks in advance.

devesh1508, 2021-03-11:
Can you please help us understand this implementation in Bullet?

erwincoumans, 2021-03-11:
1) Which Jacobians exactly? By default, Bullet and PyBullet use the PGS iterative constraint solver (projected Gauss Seidel). This solver doesn't explicitly build the Jacobian or A matrix.

2) If there are no contact points, and no other constraints (such as joint limits, joint motors etc), then there is no MLCP problem, so there is nothing to solve. If there a no contacts, but still joint limits and joint motors, then the LCP or MLCP is solved for those non-contact constraints. You will have to enable the MLCP solver instead of the iterative PGS solver.

mostafa-hussein, 2021-03-12:
Hi Erwin,
Thank you for the answer, I work with Devesh on the same problem and we are specifically looking into the second case where Bullet builds the complete MLCP and solves it.
our main point is how to force bullet to compute the LCP each time step even if there is no contact or any other constraint (I know this doesn't make much sense because there is no problem to solve ) but this will help us in our project.
1- How we can do this? 
I can be more specific we think that this parameter  "CONVEX_DISTANCE_MARGIN" is making the decision when to create the MLCP for the contact constraint or not? 
2- IS our thought is correct? 
3- we tried to change this parameter beside the safe margin parameter but it didn't change anything? Any suggestions?
#define CONVEX_DISTANCE_MARGIN btScalar(0.04)

erwincoumans, 2021-03-12:
If there is no collision detection, and no contact points there is no way to generate the LCP.

I think you mean, there IS collision detection, and there ARE contact points (with positive distance, so a positive separation instead of penetration), but Bullet is discarding those contact points?

You want to disable the discarding of contact points with positive distance, and allow to generate contact points at (large) distance?

mostafa-hussein, 2021-03-12:
Yes, that's exactly what we want to do. How we can do this and is there a function to set this distance?

erwincoumans, 2021-03-12:
Try increasing the (global) variable gContactBreakingThreshold (to allow checking collisions at a larger distance), and increase the contactProcessingThreshold (allow contacts with larger positive distance to enter the LCP). You may need to do more work, use a debugger and enter the Bullet source code, put breakpoints and see where things are discarded.

mostafa-hussein, 2021-03-12:
Great, thanks for the tips I will check these parameters.

mostafa-hussein, 2021-03-25:
Hi Erwin,
Thanks for all the help, we have a small problem if you can guide us through it will be great.
We want to add some other functions in Pybullet to expose some of the parameter and functions from Bullet that it's not available.
My idea is to do so on the C-API  (PhysicsClientC_API) and recompile the Pybullet using the  "python setup.py" but it failed.
Any help on this will be appreciated.
Thanks.

closed 2021-04-26 by erwincoumans