bulletphysics/bullet3 #3308: xarm6_with_gripper doesn't work as desired.

seolhokim, 2021-03-14:
```
import pybullet as p
import time
import pybullet_data
physicsClient = p.connect(p.GUI)
p.setAdditionalSearchPath(pybullet_data.getDataPath())
p.setGravity(0, 0, -9.8)
p.setTimeStep(1/200)
planeID = p.loadURDF("plane.urdf")
cubeStartPos = [0, 0, 0.5]
cubeStartOrientation = p.getQuaternionFromEuler([0,0,3.14/2])
robotID = p.loadURDF("xarm/xarm6_with_gripper.urdf", cubeStartPos, cubeStartOrientation)
p.setRealTimeSimulation(0)

orientation = p.getQuaternionFromEuler([3.14,0,0])

gripper_idx = 7
target_position = p.calculateInverseKinematics(robotID,gripper_idx,[0.5,0.5,0.4],targetOrientation = orientation)
p.setJointMotorControlArray(robotID,range(len(target_position)),p.POSITION_CONTROL,targetPositions = target_position)
for _ in range(300):
    p.stepSimulation()
p.getLinkState(robotID, gripper_idx)[0]

```

This result shows this arm can't calculate joint state well.

lorepieri8, 2021-03-17:
Any guesses on where the error comes from? The URDF?

erwincoumans, 2021-03-17:
You can see a working example here: 
https://github.com/erwincoumans/xArm-Python-SDK/tree/master/example/wrapper/xarm6
run 
~~~
python loadarm_sim.py
~~~
Good luck!

closed 2021-03-17 by erwincoumans

seolhokim, 2021-03-17:
What I want to ask is why gripper doesn't move to coordinates where i wanted to move correctly.

I calculated InverseKinematics and do setJointMotorControl like I did in kukagymenv.
If call kuka_iiwa/kuka_with_gripper2.sdf and do exactly same work, then It works well. but xarm6_with_gripper.urdf not.

lorepieri8, 2021-03-17:
@seolhokim 

I think you where considering the wrong links, try:

`p.setJointMotorControlArray(robotID,range(1,len(target_position)+1),p.POSITION_CONTROL,targetPositions = target_position)`

lorepieri8, 2021-03-17:
Also, maxNumIterations=50 helps in calculateInverseKinematics.

seolhokim, 2021-03-18:
@lorepieri8 It works! Thank you so much. I need to analyze it more.