bulletphysics/bullet3 #3312: Base Frame == Base COM ?

zoukj, 2021-03-17:
I found local inertia position is not zeros, when I loaded urdf of the Atlas and get dynamic infomation of the base. 
``` print(p.getDynamicsInfo(model_id, -1)) ```
the results: 
(9.509, 0.5, (0.1803825603333334, 0.13139219233333338, 0.21742011533333339), (0.0111, 0.0, 0.0271), (0.0, 0.0, 0.0, 1.0), 0.0, 0.0, 0.0, -1.0, -1.0, 2, 0.001)
1. Is the result (0.0111, 0.0, 0.0271) base com position in base frame?
2. Does that mean base com don't coincide with base frame? 
3. If base com don't coincide with frame, which is the output of *getBasePositionAndOrientation* and *getBaseVelocity*?

zoukj, 2021-03-17:
By the way, the *getDynamicsInfo()* only return **inertia diagonal**. What if it's not enough for a Irregular shape?

erwincoumans, 2021-04-26:
Yes, the base frame == CoM, and the base frame is aligned with the principal axis of inertia. This is always possible, no matter if the shape is irregular.

closed 2021-04-26 by erwincoumans