bulletphysics/bullet3 #3314: PyBullet: changeDynamics() does not work when specifying linearDamping but not contactStiffness/Damping

johannespitz, 2021-03-17:
Running changeDynamics() seems to have no effect at all when specifying linearDaming (and/or angluarDamping) but not passing contactStiffness/Damping. See the minimal example below.
```
import pybullet as p
import pybullet_data

p.connect(p.DIRECT)
p.setAdditionalSearchPath(pybullet_data.getDataPath())

cube = p.loadURDF(
    "r2d2.urdf",
    [0, 0, 0],
    globalScaling=0.1,
    useFixedBase=True,
)

for index in range(3):
    print(p.getDynamicsInfo(cube, index))

for index in range(3):
    p.changeDynamics(
        cube,
        index,
        mass=777,
        lateralFriction=888,
        spinningFriction=999,
        linearDamping=111,
        angularDamping=222,
        # contactStiffness=333,
        # contactDamping=444,
    )
for index in range(3):
    print(p.getDynamicsInfo(cube, index))

```

Output:
```
pybullet build time: Mar 12 2021 11:59:00
(10.0, 0.5, (0.0005233333333333335, 0.0033233333333333344, 0.0036066666666666686), (0.0, 0.0, -0.03), (0.0, 0.7070904020014416, 0.0, 0.7071231599922604), 0.0, 0.0, 0.0, -1.0, -1.0, 2, 0.001)
(10.0, 0.5, (0.0014166666666666672, 0.0001666666666666665, 0.0014166666666666672), (0.0, 0.0, 0.0), (0.0, 0.0, 0.0, 1.0), 0.0, 0.0, 0.0, -1.0, -1.0, 2, 0.001)
(1.0, 0.5, (3.541666525e-05, 3.541666525e-05, 2.8166665540000008e-05), (0.0, 0.0, 0.0), (0.0, 0.0, 0.0, 1.0), 0.0, 0.0, 0.0, -1.0, -1.0, 2, 0.001)
(10.0, 0.5, (0.0005233333333333335, 0.0033233333333333344, 0.0036066666666666686), (0.0, 0.0, -0.03), (0.0, 0.7070904020014416, 0.0, 0.7071231599922604), 0.0, 0.0, 0.0, -1.0, -1.0, 2, 0.001)
(10.0, 0.5, (0.0014166666666666672, 0.0001666666666666665, 0.0014166666666666672), (0.0, 0.0, 0.0), (0.0, 0.0, 0.0, 1.0), 0.0, 0.0, 0.0, -1.0, -1.0, 2, 0.001)
(1.0, 0.5, (3.541666525e-05, 3.541666525e-05, 2.8166665540000008e-05), (0.0, 0.0, 0.0), (0.0, 0.0, 0.0, 1.0), 0.0, 0.0, 0.0, -1.0, -1.0, 2, 0.001)
```

After uncommenting the lines with contactStiffness and contactDamping I get the following:
```
pybullet build time: Mar 12 2021 11:59:00
(10.0, 0.5, (0.0005233333333333335, 0.0033233333333333344, 0.0036066666666666686), (0.0, 0.0, -0.03), (0.0, 0.7070904020014416, 0.0, 0.7071231599922604), 0.0, 0.0, 0.0, -1.0, -1.0, 2, 0.001)
(10.0, 0.5, (0.0014166666666666672, 0.0001666666666666665, 0.0014166666666666672), (0.0, 0.0, 0.0), (0.0, 0.0, 0.0, 1.0), 0.0, 0.0, 0.0, -1.0, -1.0, 2, 0.001)
(1.0, 0.5, (3.541666525e-05, 3.541666525e-05, 2.8166665540000008e-05), (0.0, 0.0, 0.0), (0.0, 0.0, 0.0, 1.0), 0.0, 0.0, 0.0, -1.0, -1.0, 2, 0.001)
(777.0, 888.0, (0.04066300000000001, 0.2582230000000001, 0.28023800000000015), (0.0, 0.0, -0.03), (0.0, 0.7070904020014416, 0.0, 0.7071231599922604), 0.0, 0.0, 999.0, 444.0, 333.0, 2, 0.001)
(777.0, 888.0, (0.11007500000000003, 0.012949999999999986, 0.11007500000000003), (0.0, 0.0, 0.0), (0.0, 0.0, 0.0, 1.0), 0.0, 0.0, 999.0, 444.0, 333.0, 2, 0.001)
(777.0, 888.0, (0.02751874889925, 0.02751874889925, 0.021885499124580002), (0.0, 0.0, 0.0), (0.0, 0.0, 0.0, 1.0), 0.0, 0.0, 999.0, 444.0, 333.0, 2, 0.001)
```

I am wondering if this is intended (due to some physical limitation) or simply a bug.
Also is there any way to verify that the linearDamping and angularDamping values have been set correctly? getDynamicsInfo does not seem to return them.

erwincoumans, 2021-03-17:
Check the PyBullet quickstart guide, those values are not returned in getDynamicsInfo. 
You need to skip fixed joints, here is a test script:

~~~
import pybullet as p
import pybullet_data as pd
import time

p.connect(p.GUI)
p.setAdditionalSearchPath(pd.getDataPath())
p.setGravity(0,0,-10)

orn = p.getQuaternionFromEuler([-0.005,0,0])
p.loadURDF("plane.urdf", [0,0,0], orn)

r2d2 = p.loadURDF("r2d2.urdf",[0,0,0.5],orn)
wheels=[b"right_front_wheel_joint",b"right_back_wheel_joint",
        b"left_front_wheel_joint",b"left_back_wheel_joint"]

for j in range (p.getNumJoints(r2d2)):
  info = p.getJointInfo(r2d2,j)
  if info[1] in wheels:
    print("info[",j,"]=",info)
    #p.changeDynamics(r2d2,j,linearDamping=0, angularDamping=0, jointDamping=0)
    p.setJointMotorControl2(r2d2,j,p.VELOCITY_CONTROL,force=0)

dt = p.getPhysicsEngineParameters()["fixedTimeStep"]
print("dt=",dt)
while p.isConnected():
  p.stepSimulation()
  time.sleep(dt)
   
~~~

closed 2021-03-17 by erwincoumans

johannespitz, 2021-03-17:
Ok, getDynamicsInfo does not return linearDamping and angularDamping, but is there a function which does return those values. As fas as I can see getJointInfo doesn't return them either.

And regarding the main problem: Simply skipping fixed joints does not change anything. Replace 'range(3)' in my previous script with '[2, 3, 6, 7]' (the wheel indicies according to your example). changeDynamics does not set the mass if linearDaming is passed in as a key word argument.

erwincoumans, 2021-03-17:
No function to return the value, since you have the values, just keep track of it.

~~~
import pybullet as p
import pybullet_data as pd
import time

p.connect(p.GUI)
p.setAdditionalSearchPath(pd.getDataPath())
p.setGravity(0,0,-10)

orn = p.getQuaternionFromEuler([-0.005,0,0])
p.loadURDF("plane.urdf", [0,0,0], orn)

r2d2 = p.loadURDF("r2d2.urdf",[0,0,0.5],orn)
wheels=[b"right_front_wheel_joint",b"right_back_wheel_joint",
        b"left_front_wheel_joint",b"left_back_wheel_joint"]

for j in range (p.getNumJoints(r2d2)):
  info = p.getJointInfo(r2d2,j)
  if info[1] in wheels:
    print("info[",j,"]=",info)
    p.changeDynamics(r2d2,j,linearDamping=0, angularDamping=0, jointDamping=0, mass=99)
    print ("mass=",p.getDynamicsInfo(r2d2,j)[0])
    p.setJointMotorControl2(r2d2,j,p.VELOCITY_CONTROL,force=0)

dt = p.getPhysicsEngineParameters()["fixedTimeStep"]
print("dt=",dt)
while p.isConnected():
  p.stepSimulation()
  time.sleep(dt)

~~~
mass prints 99 for me.

johannespitz, 2021-03-17:
It turns out that passing all three damping values (linear, angular, joint) does work without contactStiffness/contactDamping.
Personally, I would still consider the behavior of my provided script as a bug (especially since it fails without telling the user), but of course I can work around the problem.

Thank you for the info and your time.