bulletphysics/bullet3 #3315: pybullet.error: getMeshData failed

shiyani21, 2021-03-18:
I am trying to use the getMeshData function for a single cube and I keeping getting error that it has failed.
here is the snippet

import pybullet as p
import time
import pybullet_data

physicsClient = p.connect(p.DIRECT)#or p.DIRECT for non-graphical version
p.setAdditionalSearchPath(pybullet_data.getDataPath()) #optionally
p.setGravity(0,0,-10)
planeId = p.loadURDF("path to plan.urdf")
cubeStartPos = [0,0,1]
cubeStartOrientation = p.getQuaternionFromEuler([0,0,0])
cube = p.loadURDF("path to cube_10_6_6.urdf",cubeStartPos, cubeStartOrientation)
print(p.getMeshData(cube, linkIndex = 0))

can anyone shed light on how to fix it.

erwincoumans, 2021-03-18:
You have to share the urdf for anyone to help, thanks. (attach a zip file)

shiyani21, 2021-03-18:
how do i do that? I am new to github

shiyani21, 2021-03-18:
[cube_10_6_6.zip](https://github.com/bulletphysics/bullet3/files/6165937/cube_10_6_6.zip)

erwincoumans, 2021-03-18:
cube_10_6_6.obj is missing, and the mtl and png/jpg texture (if any)

erwincoumans, 2021-03-18:
Since the cube has no links, you need to request the base link, which is index -1, not 0.

closed 2021-03-18 by erwincoumans

shiyani21, 2021-03-18:
I changed it to linkindex = -1, and it still failed

shiyani21, 2021-03-18:
I changed the linkIndex = -1 and it still fails

erwincoumans, 2021-03-18:
please provide cube_10_6_6.obj (zipped) and mtl file and images (if any)

shiyani21, 2021-03-18:
[cube_10_6_6_zip.zip](https://github.com/bulletphysics/bullet3/files/6166264/cube_10_6_6_zip.zip)