bulletphysics/bullet3 #3317: pybullet.error: getMeshData failed again

shiyani21, 2021-03-18:
I am trying to use the getMeshData function for a single cube and I keeping getting error that it has failed.
here is the snippet

import pybullet as p
import time
import pybullet_data

physicsClient = p.connect(p.DIRECT)#or p.DIRECT for non-graphical version
p.setAdditionalSearchPath(pybullet_data.getDataPath()) #optionally
p.setGravity(0,0,-10)
planeId = p.loadURDF("path to plan.urdf")
cubeStartPos = [0,0,1]
cubeStartOrientation = p.getQuaternionFromEuler([0,0,0])
cube = p.loadURDF("path to cube_10_6_6.urdf",cubeStartPos, cubeStartOrientation)
print(p.getMeshData(cube, linkIndex = -1)

i changed the linkIndex to -1 for the base but it still fails 
[cube_10_6_6.zip](https://github.com/bulletphysics/bullet3/files/6166134/cube_10_6_6.zip)

cadop, 2021-03-29:
Well, maybe not your problem, but your zip file is only a urdf file, there is no mesh in it.

shiyani21, 2021-03-31:
I wasn't aware you have a separate mesh file. I used blender to create the cube.

cadop, 2021-03-31:
If you look at the urdf file you gave in the zip, Line 17 is `<mesh filename="cube_10_6_6.obj" scale=".1 .1 .1"/>`. So I think if there is no `.obj` file, there is no mesh, so _possibly_ thats why you get the error.

shiyani21, 2021-03-31:
I did provide a obj file in the zip folder, if not here 
[cube_obj.zip](https://github.com/bulletphysics/bullet3/files/6238846/cube_obj.zip)

shiyani21, 2021-03-31:
I have all my mtl and obj and urdf file in the same location.

cadop, 2021-03-31:
sorry I don't get an error, but it is an empty tuple. 

From the documentation it says it is experimental and undocumented, so at least the documentation is accurate in that sense.  

I noticed that it does work on softbodies, so maybe it was a feature to help with that. 

```
import pybullet as p
import time
import pybullet_data

physicsClient = p.connect(p.GUI)
p.setGravity(0,0,-10)
cubeStartPos = [0,0,1]
cubeStartOrientation = p.getQuaternionFromEuler([0,0,0])
cube = p.loadURDF("cube_10_6_6.urdf",cubeStartPos, cubeStartOrientation, )
mesh_info = p.getMeshData(cube)
print(mesh_info)

cube_soft = p.loadSoftBody("cube_10_6_6.obj")
meshData = p.getMeshData(cube_soft)
print(meshData)
```

Running the above code for me yields:

```
?
(0, ())
(36, ((-0.6279969811439514, -0.6034669876098633, -0.5967980027198792), (-0.6237949728965759, 0.5964879989624023, -0.6062560081481934), (0.09619399905204773, 0.5939909815788269, -0.6031540036201477), (-0.6279969811439514, -0.6034669876098633, -0.5967980027198792), (0.09619399905204773, 0.5939909815788269, -0.6031540036201477), (0.09199199825525284, -0.6059640049934387, -0.5936959981918335), (-0.6331340074539185, -0.5939909815788269, 0.6031540036201477), (-0.628931999206543, 0.6059640049934387, 0.5936959981918335), (-0.6237949728965759, 0.5964879989624023, -0.6062560081481934), (-0.6331340074539185, -0.5939909815788269, 0.6031540036201477), (-0.6237949728965759, 0.5964879989624023, -0.6062560081481934), (-0.6279969811439514, -0.6034669876098633, -0.5967980027198792), (0.08685500174760818, -0.5964879989624023, 0.6062560081481934), (0.09105700254440308, 0.6034669876098633, 0.5967980027198792), (-0.628931999206543, 0.6059640049934387, 0.5936959981918335), (0.08685500174760818, -0.5964879989624023, 0.6062560081481934), (-0.628931999206543, 0.6059640049934387, 0.5936959981918335), (-0.6331340074539185, -0.5939909815788269, 0.6031540036201477), (0.09199199825525284, -0.6059640049934387, -0.5936959981918335), (0.09619399905204773, 0.5939909815788269, -0.6031540036201477), (0.09105700254440308, 0.6034669876098633, 0.5967980027198792), (0.09199199825525284, -0.6059640049934387, -0.5936959981918335), (0.09105700254440308, 0.6034669876098633, 0.5967980027198792), (0.08685500174760818, -0.5964879989624023, 0.6062560081481934), (0.09619399905204773, 0.5939909815788269, -0.6031540036201477), (-0.6237949728965759, 0.5964879989624023, -0.6062560081481934), (-0.628931999206543, 0.6059640049934387, 0.5936959981918335), (0.09619399905204773, 0.5939909815788269, -0.6031540036201477), (-0.628931999206543, 0.6059640049934387, 0.5936959981918335), (0.09105700254440308, 0.6034669876098633, 0.5967980027198792), (-0.6279969811439514, -0.6034669876098633, -0.5967980027198792), (0.09199199825525284, -0.6059640049934387, -0.5936959981918335), (0.08685500174760818, 
-0.5964879989624023, 0.6062560081481934), (-0.6279969811439514, -0.6034669876098633, -0.5967980027198792), (0.08685500174760818, -0.5964879989624023, 0.6062560081481934), (-0.6331340074539185, -0.5939909815788269, 0.6031540036201477)))
```

I'll comment again if I find anything more in the future.

shiyani21, 2021-03-31:
Thank you, I am also working on it but haven't really had any luck until your post. so, thank you tons

erwincoumans, 2021-03-31:
The feature is very limited. In a nutshell, your collision shape needs to be a mesh (obj file), not an implicit shape (cube, sphere etc). Then you need to iterate over all collision shapes. Attached is an example.
[getmesh.zip](https://github.com/bulletphysics/bullet3/files/6239663/getmesh.zip)

So generally it is best to just write your own mesh/obj parser or use assimp.

closed 2021-03-31 by erwincoumans

cadop, 2021-03-31:
@erwincoumans  this outputs the same thing I mentioned (collision shape data works, but meshinfo is empty):

```
((1, -1, 3, (1.0, 1.0, 1.0), b'', (0.0, 0.0, 0.0), (0.0, 0.0, 0.0, 1.0)),)
?
num_verts= 0
vertices= ()
```

I cloned the main repo yesterday and built with `pip install build`.