bulletphysics/bullet3 #3321: Model Matrix

shiyani21, 2021-03-21:
I am trying to do transformation from object position in the pybullet simulation to pixels and I am using projectionMatrix*ViewMatrix*ModelMatrix*Position(vector). I am not sure how to get the Model Matrix, is there a function to get the model matrix?

erwincoumans, 2021-03-21:
Each link, and the base, has a 'world transform' that you can use to build the Model matrix.

shiyani21, 2021-03-21:
The model is a plane cube and it only has a base link, so would I only use that to build the model matrix?

shiyani21, 2021-03-22:
Would there be a function like getWorldTransform?

erwincoumans, 2021-03-22:
There is pybullet.getBasePositionAndOrientation for the base, and for links there is
pybullet.getLinkState(s), see the PyBullet Quickstart Guide.
https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#

closed 2021-03-22 by erwincoumans