bulletphysics/bullet3 #3323: SystemError: <built-in function calculateInverseDynamics> returned a result with an error set

zoukj, 2021-03-24:
How to use *pybullet.calculateInverseDynamics*?
I followed <PyBullet Quickstart Guide>, but got the error. 
then, I test the *pybullet.calculateInverseDynamics* by some codes as follows, and still get the error.
```
import pybullet as p
dof = p.computeDofCount(self.model_id)
tau = p.calculateInverseDynamics(model_id, [0.]*dof, [0.]*dof, [0.]*dof)
```

erwincoumans, 2021-03-24:
There is an inverse_dynamics example here, good luck!
https://github.com/bulletphysics/bullet3/tree/master/examples/pybullet/examples

closed 2021-03-24 by erwincoumans

zoukj, 2021-03-25:
Thank you a lot! I have successfully run the example codes. 
Compare to the example, my codes load urdf model without using fixed base, and this is why I get an error. 
But how to calculate inverse dynamics for a float base model? (the *Quickstart Guide* show it's ok for a float base.)
[image]

erwincoumans, 2021-03-25:
you can try passing flags=1, see also 
https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/pdControllerStable.py#L83
used by
https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/humanoidMotionCapture.py

mrunaljsarvaiya, 2022-06-01:
> you can try passing flags=1, see also https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/pdControllerStable.py#L83 used by https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/humanoidMotionCapture.py

What does flags=1 do?

Senwei-Huang, 2023-03-23:
> you can try passing flags=1, see also https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/pdControllerStable.py#L83 used by https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/humanoidMotionCapture.py

I encountered the same problem. This solution cannot solve my problem. The error is reported as follows:
```
  File "E:\VirtualMachine\OneLeg\pdControllerStable.py", line 138, in computePD
    G = self._pb.calculateInverseDynamics(bodyUniqueId, q1, qdot1, zeroAccelerations)
SystemError: <built-in function calculateInverseDynamics> returned a result with an error set
b3Printf: b3Warning[examples/SharedMemory/PhysicsClientSharedMemory.cpp,1083]:

b3Printf: Inverse Dynamics computations failed
```
How can I solve it? Could you give me a hand?
@erwincoumans

Senwei-Huang, 2023-03-26:
Thank you. I solved this problem by myself. The reason for the problem is that I did not consider the degree of freedom of the floating base. However, I am not sure what specific changes have occurred after the addition of the `flags `keyword, as I did not find a description of the `flags `keyword in the help documentation. Can you provide further explanation for this keyword? Thank you very much in advance.

> I encountered the same problem. This solution cannot solve my problem. The error is reported as follows:
>
> ```
>   File "E:\VirtualMachine\OneLeg\pdControllerStable.py", line 138, in computePD
>     G = [...]